prusa3d/PrusaSlicer #9615: shelfing

comemilg, 2023-02-05:
### Description of the bug

my problem is that some of my 3d prints just jump one layer randomly. some prints don't jump layers and other prints jump on the first layer. an if I try to print it with the same gcode it happens at exactly the same place. everything I print I design on fusion 360. if I don't change anything in Prucha slicer but I change for example the height it prints just fine then I do something else and it starts to shelf. and I don't know what's the problem is. it's often bigger prints it happens to.

### Project file & How to reproduce

[image]

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2,5,0

### Operating system

windows 11

### Printer model

cz-300 Ctazy3DPrinter

gudvinr, 2023-02-07:
That probably has nothing to do with the slicer.  
Check belt tension and maybe lower acceleration in the direction where you see layer shifts.

comemilg, 2023-02-07:
I have tightened all my belts, so I don't think that's the problem. The layer shifts happen in the exact same place if I print the same part. But if I print something else or modify the part in fusion 360 the shelfing doesn't happen. So it must have something to do with the software. And it happens in the Z direction every time.

foreachthing, 2023-02-07:
I had once the same issue with the UM2. The belts were tight, only the driving wheels (geard wheel on connecting rod and stepper) were loose. After tigheting the setscrews it was solved.

comemilg, 2023-02-07:
Did it happen in the same location every time? that's the part I find weird.

foreachthing, 2023-02-08:
> Did it happen in the same location every time? that's the part I find weird.

On the same part, yes, pretty much. I guess the acceleration was the same at the same location.
I then slightly rattled the hotend and noticed that the stepper wasn't moving....

This cannot be a software issue. Try a different slicer to confirm. Try Slic3r, CraftWare, Cura, or any other free slicer.

lukasmatena, 2023-02-08:
I agree with @foreachthing that this is quite likely a problem in the configuration / hw / both. Either way, you did not provide any data. We would need to see the 3MF and G-Code - with the exact location where you think the problem is. Otherwise, there is not much to discuss.

comemilg, 2023-02-10:
[K2_Clip_Locking.zip](https://github.com/prusa3d/PrusaSlicer/files/10704700/K2_Clip_Locking.zip)
i have downloaded this file form printeble.com and just sliced it an printed it. and this happend  
[image]
dose the possibility that the sdcard is bad.

lukasmatena, 2023-02-10:
This is certainly not slicer related. I can see that you are using default Print, Filament and Printer profiles without a single modification, so it is not surprising it does not work. Maybe the acceleration the printer is aiming for is too much, maybe something else is misconfigured. Closing.

closed 2023-02-10 by lukasmatena