prusa3d/PrusaSlicer #9616: chyba slicerování

HEKRcz, 2023-02-05:
### Description of the bug

Zdravím, prusaslicer 2.5.6 a2 mi při vybrání objektu který máte v seznamu (jehlan / pyramida) a zvetšení na 30x30x30 po Slicerování vyhodil text... PrusaSlicer narazil na fatální chybu: "Porušení přístupu"
Uložte projekt a restartujte PrusaSlicer. Budeme rádi, když nám problém nahlásíte.
zasílám obrázek před Slicerováním.
Děkuji .

[image]

### Project file & How to reproduce

nic vic, jen ve vasem seznamu přidat tvar "galerie" vas objekt a slicerovat, udela to pro vsechny modely v galerce, (kostka, valec, koule bez problemu

### Checklist of files included above

- [ ] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.6.0a2-win64

### Operating system

w10

### Printer model

creality ender 5

rtyr, 2023-02-05:
Můžete prosím zkusit s aktuální verzí alpha3? Případně přidat 3MF projekt.

HEKRcz, 2023-02-06:
Zdravím, tak na 2.6a3 bez problému, muselo to být něco ve 2ce.

rtyr, 2023-02-07:
OK. Díky za vyzkoušení. Zavírám issue.

closed 2023-02-07 by rtyr