prusa3d/PrusaSlicer #9619: Allow temp control for print moves

D3MZ, 2023-02-05:
**Is your feature request related to a problem? Please describe.**
My bridges and non-print moves are stringy/saggy w/ PETG.

**Describe the solution you'd like**

Allowing temp control for moves, like:
1. Cold bridges (Reduced sagging)
2. Cold non-print moves (Reduced stringing)
3. Hot infill (increased layer adhesion)

1. Change Print Settings -> Speed to Print Settings -> Speed And Temperature
2. Allow user to set temps for a given print move, along with speed.

**Describe how it would work**
Add a Temp column under the Print Settings -> Speed UI.

**Describe alternatives you've considered**
Unfortunately, this is the only way I can think of that'll allow for having both flat bridges and strong layer adhesion.

rtyr, 2023-02-07:
I believe you can already create such g-code using g-code substitutions feature (M104/109 based on print features).

That being said, I am afraid this is not realistic approach because actual temperature change takes very long time in reality and waiting xx seconds before/after some features is also not an option.

So I think using g-code substitutions should be more than enough to cover your request.

k1-801, 2023-02-09:
I actually disagree with both statements. I did make a similar issue, #9523, and described it in far more detail.
In short: on a typical machine it takes around 15 to 30 seconds to drop the temperature to the bridgeable temperature (a drop of 10-20 degrees C), which *does* require to move the nozzle away from the part, but *is* affordable in print time, as long as you have a reasonably low number of layers with bridges in them. Since bridges require cooling anyway, it would be beneficial to turn the fan on before using the "wait for temperature" command, so the fan helps to cool down the nozzle. It should be kept on until the bridge is done, as not to wait for it to spin up *while printing*.

rtyr, 2023-02-09:
> as long as you have a reasonably low number of layers with bridges in them

Which is rarely the case unless you print one very simple test object. Bridging is used also on perimeters and over sparse infill.

> which does require to move the nozzle away from the part

And that is a problem. Since you would need to print some sort of tower to prime the nozzle properly before the return. You cannot expect that there will be no leak in 15-30 seconds

k1-801, 2023-02-09:
> Bridging is used also on perimeters
[image]

By the way, bridge settings seem to not apply to perimeters. You can see them sag on this picture, where the regular bridges don't.

rtyr, 2023-02-09:
> By the way, bridge settings seem to not apply to perimeters.

depends on settings in general (detect bridging perimeters). Also note there is big difference between thick bridges and non-thick bridges (thick bridges parameter). But let's not clutter this issue with different topics.

k1-801, 2023-02-19:
> And that is a problem. Since you would need to print some sort of tower to prime the nozzle properly before the return. You cannot expect that there will be no leak in 15-30 seconds

Did some experiments. ABS, 250 degrees (standard settings I print at) gave less than a millimeter long drip on the nozzle after idling for 30 seconds, even without dropping the temperature simultaneously, then it almost stopped growing. Almost. As long as this drop ends up anywhere inside the infill (heck, even if it gets wiped over a perimeter), I'd be fine with it. As per underextruding when the bridge starts, it also doesn't sound like too much volume to care. We also have a setting "wipe on infill", which allows to solve exactly that problem when no tower is present.

Did not run the same test on PLA or TPU, though, the situation might be different.

rtyr, 2023-02-20:
I suggest you to use g-code substitutions, it should allow you to do it. You will see that it is not good idea to wait somewhere and then return and start printing something with heavily under-pressurized nozzle.

closed 2026-09-29 by prusaslicer-bot[bot]