prusa3d/PrusaSlicer #9620: Both extruders heat up on second layer in Dual Extruder system

3DSmitty, 2023-02-06:
### Description of the bug

I have a dual extruder system. All works good except after the first layer both extruders heat up and the second extruder oozes a little causing artifacts in the print. I assume only the active extruder (T0) should be heated not both of them. Is this a bug in prusaslicer or am I missing something?

Gcode:
;LAYER_CHANGE
;Z:0.44
;HEIGHT:0.2
G1 E30.84135 F6000
G92 E0
G1 Z1.24 F9000
M104 S235 T0 ; set temperature
M104 S235 T1 ; set temperature
M140 S80 ; set bed temperature
G1 X121.039 Y128.961
G1 Z.44
G1 E10 F6000
;TYPE:Perimeter

### Project file & How to reproduce

[xyzCalibration_cube13.txt](https://github.com/prusa3d/PrusaSlicer/files/10637292/xyzCalibration_cube13.txt)

### Checklist of files included above

- [X] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.5.0+win64

### Operating system

Windows 10

### Printer model

Geeetech A20M with dual extruder conversion

lukasmatena, 2023-02-06:
@3DSmitty It is not a bug in the sense it would do something different then intended. This is how Slic3r and PrusaSlicer behaved for a long time.

However, there is now a new alpha version [2.6.0-alpha3](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.6.0-alpha3) which changes the behavior with dual extruder setups (described [here](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.6.0-alpha2)). I would be glad if you could try it and give us some feedback about how the changes affected your particular setup. Namely, this problem should be possible to solve by using the reworked "Ooze prevention" function.

3DSmitty, 2023-02-06:
Those changes sound great! I will try it out and report back, thank you.

closed 2026-09-29 by prusaslicer-bot[bot]