prusa3d/PrusaSlicer #9621: Support Enforcers dramatically over-supports with Organic Supports

rcmaniac25, 2023-02-06:
### Description of the bug

Prior to organic supports, in an effort to have more efficient supports, I would slice a model, check where overhangs/islands were, and then paint supports in those areas. With 2.6, my plan was to take those projects and simply switch from snug supports to organic supports and reap the benefits. In the release notes I saw there as a new support painter that is supposed to take "many factors such as the center of mass, weight of the supported part, bed movement, potential extruder collision, bed adhesion, material, and bridging." into account. So I opted to try that out on some non-supported models, and I thought it operated similar to how I would (doing single circles on spots, while painting some others).

Problem:
The models were nearly wrapped in supports. To the point where I wasn't sure I'd get models out of the supports... especially delicate ones that would really benefit from organic supports. It also took a very long time to generate supports... I literally made dinner one night (~30 min) waiting for one model to finish getting supported.

They were even worse then snug and grid supports:
- grid: 3h 39m, 5.60m filament
- snug: 3h 12m, 4.82m filament
- organic: 3h 40m, 5.56m filament

Expected:
Support enforcers would simply limit where support contact points would be generated and, like existing supports, if there was no actual area that could be supported it didn't... or at worst it would produce supports "near" the model without making any contact. The supports would be little pin-prick like supports like the pics everyone (Prusa included) has. And they would actually be less material and print time then non-Organic supports (regardless of support enforcers or not).

What I was told:
After much frustration, I saw person after person (Twitter, MakerDeck, etc.) who used organic supports get nice little pin-prick supports instead of a mass of supports I was getting. I asked for help. I was told by everyone that they didn't use support enforcers and simply chose "everywhere" for supports. They simply set to Organic and went about there day. No automatic support painter. No change in separation distance, interface layers, etc. No "avoid crossing curled overhangs". Nothing. Just "Everywhere" supports of type "Organic".

So to me, either support enforcers are not supported yet, organic supports is trying to support every inch of support enforcers, or the recommendations for the alpha don't actually work.

Some numbers (repeated part from above)

- No supports (would fail to print): 2h 35m, 4.08m filament
- Everywhere (grid): 4h 23m, 7.08m filament
- Everywhere (snug): 3h 39m, 5.90m filament
- Everywhere (organic): 3h 11m, 4.84m filament
- Support Enforcers (grid): 3h 39m, 5.60m filament
- Support Enforcers (snug): 3h 12m, 4.82m filament
- Support Enforcers (organic): 3h 40m, 5.56m filament

Screenshots

Everywhere (slice time: 12.68 sec)
[image]

Support Enforcers (slice time: 60.58 sec)
[image]

### Project file & How to reproduce

Reproduce:

1. Import file (I used a Prusa MK3S+ and 0.15mm quality settings and Prusament PLA)
2. Turn on support enforcers
3. Per instructions on alpha2's page (since alpha3 doesn't have separate instructions)
    1. Turn on "Avoid crossing curled overhangs"
    2. Change support style to "Organic"
    3. Change "top contact z distance" to a multiple of layer height, so 0.15
    4. Turn off "top interface layers" by setting it to 0
5. Open paint on supports and choose "automatic painting" (I also painted 2 spots under the tail, but issue occurs without this)
6. Slice

Projects

2x 3mf in zip file:

- Everywhere: model with "Everywhere" support setting
- SupportEnforcers: model with "For support enforcers only" support setting

[Projects.zip](https://github.com/prusa3d/PrusaSlicer/files/10660425/Projects.zip)

Model: https://www.printables.com/model/359372-couatl

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

Version 2.6.0-alpha3+win64

### Operating system

Windows 10 Pro 21H2

### Printer model

Prusa i3 MK3S+

jasonleach-dev, 2023-02-07:
I had the same behaviour/concerns - default 'everywhere' wasn't quite accurate enough, and painted supports were far too overdone (both manual and automatic painting).

Adjusting the advanced tree support settings branch diameter to 1.5mm and tip diameter to 0.2mm (using a 0.2mm nozzle) helped reduce the over-supporting, and further tweaking of parameters can likely improve it further.

DerGenaue, 2023-02-27:
Support Enforcers have code that compensated for this bug over here: https://github.com/prusa3d/PrusaSlicer/issues/9459 / https://github.com/prusa3d/PrusaSlicer/issues/9556
The bug was fixed in Alpha 4, but it might be that the workaround now makes it generate even more supports.

Organic supports are intended to also support the sides of prints when used with enforcers afaik.
Basically to stabilize tall prints on their sides so they don't wobble.

rcmaniac25, 2023-02-28:
@DerGenaue I'm not sure about that. With non-organic supports, there are some structural elements (like needing to bridge a gap between the zig-zag of the support structure) that may need to extend past the sides of the model. But otherwise it gets near the support area, makes an interface, and then stops.

In alpha2's release notes it said that interface layers don't work right now. I don't know if the plan is to make them work or if the plan is they won't be used/needed. And assuming you came over from my comment from #9801, I can't change overhang threshold with support enforcers to get a better result. So it makes support enforcers not very useful as of now.

Support enforcers used to be "reduce the amount of supports and only support the parts that need them". This makes them not useful with organic supports. If that's the case, then it's a repeat of #9801 of too many supports in by default.

closed 2026-09-29 by prusaslicer-bot[bot]