prusa3d/PrusaSlicer #9624: Please Add Monotonic Lines Sparse/Solid/Bridge Infill (2.6.0)

limaciform, 2023-02-06:
**Is your feature request related to a problem? Please describe.**
I did search and couldn't find anything else related to the 'Monotonic Lines' infill. And I may be completely jumping the gun on something that's already being worked on. But I figure it can't hurt to ask anyway.

Connected/anchored rectilinear infill has always had problems with the extruded sections that run parallel to the perimeters. You either have to slightly overlap these sections with the innermost perimeter and accept the holes left along the surface edge, or you have to increase the overlap and accept the over extruded mess.

[image]

**Describe the solution you'd like**
_Top/Bottom_
The new 'Monotonic Lines' infill for the top/bottom pattern is a great addition, and the perfect solution to this problem for the top/bottom layers.

_Sparse Infill_
Selecting 'Rectilinear' for the sparse infill, and setting 'no anchor' almost changes the sparse infill to be ideal, but the path sequence is not monotonic as would be preferred. Monotonic paths are preferred because as the nozzle travels along the innermost perimeter to the next infill line, it will help to iron the infill lines to the perimeter, and any oozed material will act like anchoring without over extruding. Adding a 'Monotonic Lines' infill pattern for sparse infill would solve this.

_Solid Infill_
Unfortunately no settings change this at the moment, meaning we're forced to use rectilinear infill, and left with holes or over extrusion. An option to choose the solid infill pattern, with a 'Monotonic Lines' option would fix this.

_Bridge Infill_
Unfortunately no settings change this at the moment either. I find that lines with a 50% or so overlap tend to anchor better for bridging than forcing the rectilinear anchor sections to overlap the innermost perimeter and making an mess. An option to choose the bridge infill pattern, with a 'Monotonic Lines' option would fix this.

**Describe how it would work**
SuperSlicer has these settings, so it's a good reference for this feature.

[image]

**Describe alternatives you've considered**
Changing to SuperSlicer, but unfortunately each has features the other doesn't. It would be great to see more collaboration.

**Additional context**
[image]

MooMinIL, 2023-02-06:
This is a great point.
I also find that having no option for changing the Bridge Infill can lead to some pretty bad looking bottom overhangs - especially when creating an embossed pattern facing the build plate.

limaciform, 2023-02-07:
> 

Definitely agree with you there. The way those anchors/connections get squashed against the innermost perimeter while printing over open space, creates a horrible looking finish, and a weak point. 

[image]

closed 2026-09-29 by prusaslicer-bot[bot]