prusa3d/PrusaSlicer #9625: Wipe Tower doesn't support different extruders for support

Agent-Cain, 2023-02-06:
### Description of the bug

I'm not sure if this is the right channel to report this problem, if not, please let me know where it would be appropriate.

I've seen the picture on the startup screen with the benchy where support was printed in a different filament.
[image]
While trying to recreate something similar I run in this issue: 
[image]

If this benchy was printed on the mini, as the picture implies, how did you get the filament purge managed ? 
What am I missing ? 

Thanks in advance. 

### Project file & How to reproduce

[Hulk Bust.zip](https://github.com/prusa3d/PrusaSlicer/files/10661583/Hulk.Bust.zip)

### Checklist of files included above

- [X] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.6.0-alpha3

### Operating system

Win10

### Printer model

Hypercube - dual extruder - single nozzle

rtyr, 2023-02-06:
If you need wipe tower, you can currently print it like that only if you set contact Z distance to 0.

Agent-Cain, 2023-02-06:
I swear I tried it :D 
But it works now, couldn't reproduce toe error. 
Thank you !

closed 2023-02-06 by Agent-Cain