prusa3d/PrusaSlicer #9626: Add possibility to use differents type of support in a single model

cirokk, 2023-02-06:
The possibility will be incredible because there is a model that has parts that work better with organic support and parts that work better with grid support.

Under custom supports, add a selection of what type of support we want to paint. When we paint a custom support, the color is blue, so set blue for grid, green for organic, and yellow for Snug. to use in a single model

foreachthing, 2023-02-06:
I was just thinking the same thing!! 
One section of the model could use "regular" support and the rest would be organic support.
Or, if a modifier could accept "support type"; that would help me out.

mix579, 2023-02-06:
Sounds like a good idea.

Tupson444, 2023-02-06:
This was already suggested here: #9553

closed 2023-02-08 by bubnikv