prusa3d/PrusaSlicer #9628: MacOS - GUI font and items way too small

TomMuc1, 2023-02-06:
### Description of the bug

the fonts and elements of the graphical user interface in macos on newer notebooks and also external 4k monitors is much too small and therefore completely unusable.

here were already several hints/threads about it - exactly nothing happened - that's why here again new.

and no - this is not a question of age or any other settings in the operating system because the user interfaces of ALL other applications already work wonderfully in the basic settings. many applications also have the possibility to customize the gui elements (fonts icons etc).

there is no need for further explanations - just have a look at it on a current mac! and if you don't understand that, then you should stop designing graphical user interfaces as a programmer.

and - YES - this is a bug - a huge one.

cheers
tom
[image]

### Project file & How to reproduce

not needed

### Checklist of files included above

- [ ] Project file
- [X] Screenshot

### Version of PrusaSlicer

any

### Operating system

macos

### Printer model

any

mpaw, 2023-02-26:
@TomMuc1 I have been using PrusaSlicer on my 16" M1 Pro MacBook Pro and two external 4k monitors and the font size has been fine. Maybe you need to modify your display scaling in settings. I think if the text was any larger it would be difficult to fit all the information and options that the slicer gives you.

[image]

TomMuc1, 2023-02-26:
hello marcin,

first of all thank you for all your effort and also with the screenshot! but - yes - we work here on mac and pc with all adobe products and 3d programs like solidworks etc. and of course we also are aware of these settings.

but the screenshot doesn't show it quite right. here are two where you can judge it better. there you can see that both apple and adobe use this small type only for subordinate help texts. everything relevant is larger and/or bold. and you can also see that with adobe you could make the fonts and icons larger but we have also here only the standard size 'small' set.

as i said - we don't have this problem with any software except prusa slicer. but no matter we don't run after anyone. if they and mr. prusa don't like it they should leave it. the market will regulate it. and when someday everyone will use cura and creality or bambu printers they'll wake up. too late ...

[image]

[image]

mpaw, 2023-02-26:
Hi @TomMuc1 I actually run the scaling on the macbook on the same setting as you

[image]

On my previous screenshot it was the scaling from the external monitor. It's just the external monitors that have the scaling larger as they are big monitors and window size matches the macbook display at that scale.

I'm also confused by your Adobe screenshot? Isn't all the text here either the same size or even smaller than PrusaSlicer?

[image]

TomMuc1, 2023-02-26:
nope - it's smaller and in addition they use less contrast between background and fontcolor. but as i said - i am done with all these unprofessional coders having zero clue about ui design.

[image]

Jan-Soustruznik, 2024-03-12:
Closed as duplicate of https://github.com/prusa3d/PrusaSlicer/issues/6317

closed 2024-03-12 by Jan-Soustruznik