prusa3d/PrusaSlicer #9629: Cut tool will not cut unless there is an object

gloverd, 2023-02-06:
### Description of the bug

Cut tool will not work to separate objects if the cut is performed in the space between them. The "Split to Objects" command works for this, but if a group of objects want to be separated into 2 or more groups instead of many individual objects the cut tool used to be usable. 

Screenshot from 2.5.0 where such a cut is allowed 
[image]

Screenshot from 2.6.0 [image]

### Project file & How to reproduce

[Cut Middle.zip](https://github.com/prusa3d/PrusaSlicer/files/10664742/Cut.Middle.zip)

1. Import a model or cut a model so there are two separate bodies inside of it
2. Use the cut tool to cut along the separation.
3. "Perform Cut" button will be faded and unable to work in 2.6.0-alpha2
4.  Screenshot from 2.6.0 [image]
5. Screenshot from 2.5.0 where such a cut is allowed 
[image]

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

Version 2.6.0-alpha2+win64

### Operating system

Windows 10

### Printer model

Prusa MINI

YuSanka, 2023-02-09:
Duplicate of #9605
And a same question for you: Why do you use cut tool instead of "Split to Objects" functionality?

FidelCapo, 2023-02-16:
Thanks, for your issue. We enabled this option again. Fix will be part of PrusaSlicer 2.6.0-alpha4.
Closing this issue.
SPE-1466.

closed 2023-02-16 by FidelCapo