prusa3d/PrusaSlicer #9632: Adding connectors to either split cut part

3DPrintingGeek, 2023-02-06:
**Is your feature request related to a problem? Please describe.**
Currently the "Add/Edit connectors" feature seems only allowing to add connectors to the lower part of the cut, which is a little bit annoying, when that part has a flat surface, which makes it a top candidate for laying flat on the buildplate, while one wants to print the remaining part in an angle with supports. If one doesn't want to use dowel connectors,  this is not possible anymore.

**Describe the solution you'd like**
It would be great, if one could choose, to which of the sliced parts to add the connectors to.

**Describe how it would work**
Radiobuttons or a combobox to choose the part (upper / lower) to apply the connectors to.

**Describe alternatives you've considered**
Currently the only way to work around is to flip the object by 180° on the X or Y axis before using the cut tool.

bubnikv, 2023-02-07:
I suppose flipping the plane orientation would be sufficient, right?

3DPrintingGeek, 2023-02-07:
> I suppose flipping the plane orientation would be sufficient, right?

You're correct.  I hadn't thought of that before. I tried that and it works! That's even better than flipping the object as I did to get there.

kubispe1, 2023-02-16:
Flipping of the cut plane is implemented and will be part of the next release.
SPE-1464
commit: b40473be51fd1050fa4d7b2c3297230758085d46

closed 2023-02-16 by kubispe1