prusa3d/PrusaSlicer #9633: Add option to Invalidate Cut Info at the time of cutting

Snuff1eupagus, 2023-02-06:
**Is your feature request related to a problem? Please describe.**

Many times we know when we preform a cut that its all we needed to do, the way its set up in the current alpha versions we have extra steps to preform after making the initial cut. If making several cuts this can become cumbersome. 

**Describe the solution you'd like**

Add a radio button that we can tick to the cut tool dialog. if selected it would negate the need to Invalidate Cut Info after each cut.

**Describe how it would work**
If you already knew that you wanted to commit to the cut, you would tick the button, saving all the steps of invalidating the cut info.
If we still wanted the cut info we could decide not to tick the button and have the option to move forward with the new features. 

[image]

DragonFinlay, 2023-02-07:
While I understand what you mean, I think a compromise between how it is now and what you're suggesting would be best.
We have #9520 which suggests at least saving all the settings that were entered when using the tool last time.

While your suggestion would be just one click (taking away some control), the solution above would mean two clicks per cut while still preserving full control.

What are your thoughts?

Snuff1eupagus, 2023-02-13:
@DragonFinlay 

Currently the cut tool saves your last selections for your "After cut:" choices on a per session basis, and I think that's good for this tool. Currently in the alpha, If I make a cut with the tool and save both pieces, I have to right click on one of the parts as an added step, to "Invalidate the cut info" The cut info is left in an open editable state, to allow for the addition of dowels and plugs. I won't be using this functionality of the cut tool, and would like a radio button that I can select , at the very least on a per session basis to negate the need for this process. If I'm making many cuts on several objects, as I often do, this becomes very tedious.  

I think it's fine to have the "Invalidate the cut info" in the popup menus for the objects, for folks who want to use the plug and dowel features.

I'm not looking to take that control away, I'm looking for added control, in the form of a choice. Meaning I can decide to "Invalidate the cut info" a the time of cut or leave it open for further editing, and it's good for  everyone. 

I looked at the #9520 suggestion and don't find that we are asking for the same thing.

What do you think?

kubispe1, 2023-02-16:
We implemented this issue. Will be part of the next release.

closed 2023-02-16 by kubispe1

YuSanka, 2023-02-16:
> We implemented this issue. Will be part of the next release.

Small note: The new implementation will have different behavior. 
If no connector has been added to the cut, a cut information will not be applied to the cut.

Snuff1eupagus, 2023-02-16:
Thank you for your efforts.