prusa3d/PrusaSlicer #9636: PrusaSlicer crashes when saving a project

APEbbers, 2023-02-06:
### Description of the bug

When save a project prusa crashes with an internal error.
[log.txt](https://github.com/prusa3d/PrusaSlicer/files/10668566/log.txt)

### Project file & How to reproduce

Open project->Save project.
Happens random. But files are in onedrive

[Casing - parts - Casing.zip](https://github.com/prusa3d/PrusaSlicer/files/10668596/Casing.-.parts.-.Casing.zip)

### Checklist of files included above

- [X] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.6.0-alpha-3

### Operating system

Windows 11

### Printer model

Ender 3 with skr 1.2 mainboard and dual extruders

kubispe1, 2023-02-06:
And When you first discovered this ugly behavior ? Only in 2.6.0-alpha-3 or you had this same issue in 2.5.0? Can you try it? It is difficult to reproduce...

APEbbers, 2023-02-06:
The first time was with 2.6.0-alpha-3. Previous versions are ok.

 �

 �

closed 2026-09-29 by prusaslicer-bot[bot]