prusa3d/PrusaSlicer #9637: Max Print Height - Printer setting

chinoco02, 2023-02-06:
Hello PS github community,

I have a custom printer and I have to print a model that is much taller than the current 1200mm limit.

Is there any chance to increase the max print height setting under the Size and Coordinates section in the Printer settings tab?  I prefer to have the number be entered in by the user by default.

Superslicer has this feature, but their Arachne version is not as mature as Prusaslicer 2.5/2.6.

Thanks!

rtyr, 2023-02-07:
I am closing this as a duplicate of https://github.com/prusa3d/PrusaSlicer/issues/6072

closed 2023-02-07 by rtyr

bubnikv, 2023-02-07:
>I have a custom printer and I have to print a model that is much taller than the current 1200mm limit.
What are your printer dimensions?
You may consider changing scaling of your firmware to centimeters and then to work in centimeters in PrusaSlicer.

chinoco02, 2023-02-08:
Hi Vojtech,

Many thanks for the help. How would I go about changing the scaling in the
Klipper firmware?  And how do I change the units to centimeters in
PrusaSlicer?

Kind regards,