prusa3d/PrusaSlicer #9638: Excessive CPU usage when PrusaSlicer Idle

SockedIn, 2023-02-06:
### Description of the bug

If I leave PrusaSlicer open in the background it gradually uses more and more CPU resources.  I've seen it as high at 76%.

### Project file & How to reproduce

Drag .stl file onto PrusaSlicer.
Rotate part appropriately
Add text - negative setting
Rotate part to best printing orientation
Slice
Export to Octoprint (OctoPi running on my Ender 3 V2)
Print.  
Web browsing or watching YouTube videos.
After about 30 minutes I notice fans are on and when I check CPU usage very high.
Close PrusaSlicer and fans stop and CPU usage is obviously gone.
[image]
[image]

### Checklist of files included above

- [ ] Project file
- [X] Screenshot

### Version of PrusaSlicer

Version 2.6.0-alpha3+MacOS-x64

### Operating system

macOS 13.0.1 (22A400)

### Printer model

Creality Ender 3 V2

gudvinr, 2023-02-07:
It might be similar to what I had in #8871 but it was on Linux. Is it only happening in gcode viewer?

SockedIn, 2023-02-10:
It was happening just when viewing the sliced model.  I was able to reproduce it several times, but today I can't make it happen.  I'll watch for a day or two and close this if it is no longer happening.

SockedIn, 2023-02-10:
Had PrusaSlicer open in background with a small project and noted the increased CPU usage again (46%).  It happened in the sliced view.  When I switched to the plater view the CPU usage dropped to zero.

sbierly, 2023-07-19:
I was minding my business checking email this morning, and wondered why fans were whining away on my laptop. Turns out Prusaslicer (2.6.0), idle in the background from slicing something last night, was using 35% of the CPU, continuously (I had the graph open), the laptop very hot. What the heck? The program should be doing nothing, acting more like malware...

adrianVmariano, 2023-10-01:
I see this happening in 2.6.1 where it will CPU use will rise to 1300% (as reported in top, 13 cores fully loaded) running under linux.  It will load the CPU like this for maybe ten seconds and then go quiet again.

mdeeter, 2023-10-05:
I'm also seeing this.  Occasionally, things seem to slow down... I check Activity Monitor and see Prusa Slicer sitting about 70-90% CPU... but it's just idle in the background.

Next time I see this, I'll try changing to platter view and see if it drops.

MacBook Air (M2) 15inch 2023
24GB ram

mdeeter, 2023-10-07:
Happened again... kicked off a print and left Prusa Slicer running the background.  An hour later, I'm wondering why the laptop is getting so warm.

[image]

It was sitting in the bg like this:
[image]

After pulling PrusaSlicer to the foreground and without making any changes or clicking on any UI elements in the app, I checked the Activity Monitor again... it dropped to 3%... and then a few minutes later:
[image]

MacBook Air (M2) 15inch 2023
24GB ram
PrusaSlicer 2.6.1 MacOS-arm64

David-tech42, 2024-02-24:
On Win10-64 I get slightly different behavior using PrusaSlicer 2.7.1. but same effect (laptop getting warm because of CPU usage)
1. I check CPU usage and process usage before starting anything ("System" process - ntoskrnl.exe value is low or insignificant).
2.Then I only launch PrusaSlicer (but do nothing, even can minimize it), then
while prusaslicer process usage stay low the "System" ntoskrnl.exe process consistently rise to 12-15% until I close PrusaSlicer.

after few minutes, laptop get hot and fans start to kick in.
looks like prusaSlicer is triggering some kind of active poll in kernel....

if this can help debugging...

mdeeter, 2024-02-25:
I resolved this by switching to OrcaSlicer.

bathrobehero, 2024-02-27:
Here's to reproduce:

 - open QIDISlicer (no need to open any project), cpu usage is 0
 - click on the devices tab
 - click on any other tab and CPU usage goes high and stays there.
 
Weird thing is if you click on the device tab again, CPU usage goes down to 0, until you click on any other tab again. If you never click on the devices tab, the issue never occurs.

slashbeast, 2024-09-06:
I can confirm that it only happens when I am in preview and going back to the first view screen solves the elevated CPU usage. MacOS, M3 Pro CPU.

ghostface, 2025-03-01:
> I can confirm that it only happens when I am in preview and going back to the first view screen solves the elevated CPU usage. MacOS, M3 Pro CPU.

Same exact thing here

magore, 2025-09-26:
Still happens in 2.9.3 getting constant 30% CPU when idle and computer is slow to respond

closed 2026-09-29 by prusaslicer-bot[bot]