prusa3d/PrusaSlicer #9640: Avoid printed parts and supports when traveling

lucasavendano, 2023-02-07:
I think it´s time to solve the only problem that doesn´t have a solution since "always?".

Don´t get me wrong, PS is exceptional, I stopped using cura at all since PS. But some materials don´t stop leaking despite hardware, software or new great features (thanks for all that).

The "avoid crossing perimeters" feature is not effective, it is necessary that when traveling, not only printed parts and supports are avoided, but also the area that the part or the support will occupy must be avoided. I mean, before and after printing.

The next picture shows how are we now, there is not much to do... 
[image]

And this is how Cura manage the same case:
[image]

I don´t like to demand something that i can´t do by myself but the truth is i can´t.

If can do something to solve this, i think PS will be the perfect slicer, for me it is the best "."

I used PS for more than 2 years, it slices very fast, I think there wouldn´t be problem if it takes a little more to slice with better travels, If the problem is the time of course, there could be other problems to solve, i don´t know. How fast is the slice is useless when the printed part is useless.

Hope my request will be usefull for every one! Thank you PS!

Pictures that shows better the problem:
[image]

[image]

foreachthing, 2023-02-07:
PS has this option here: https://help.prusa3d.com/article/layers-and-perimeters_1748#avoid-crossing-perimeters
But it's not very efficient to say the least. With some prints, it travels around the part, just to print some speck of infill and then travels aaaaall the way back to _start_ the outer perimeter (when it could have started right where it was). And the value has to be high otherwise it would just travel right through everything - like in the images above.
Long story short: path planning is not in the _strength_ column of PS :-\

Sousuke-Cheng, 2023-05-20:
I also encountered this problem. It keeps going over my support back and forth and eventually crushes on one of them. I was using the new organic support, dunno if it will be fixed when releasing... or I have to go back to cura...

GabrielCJS, 2023-08-31:
Same thing here. My nozzle keeps knocking organic supports down. On time it knocked it down and pushed it against the rest of the part blocking the movement of the extruder to the point my belt jumped some teeth. I tried slicing the same file with and without the "avoid crossing perimeters" turned on, but it seems to make little to no difference.

albert9052, 2023-09-01:
I raise the nozzle (z-index) after printing every small part to kind of resolving this issue, I think it's something in the retraction part of the setting, but I'm not sure 🤔

jstaubr, 2023-09-01:
> I raise the nozzle (z-index) after printing every small part to kind of resolving this issue, I think it's something in the retraction part of the setting, but I'm not sure 🤔

Yes, lift Z can be a workaround but creates others issues in print like strings and wisps which cannot be mitigated with retraction setting.

GabrielCJS, 2023-09-01:
> Yes, lift Z can be a workaround but creates others issues in print like strings and wisps which cannot be mitigated with retraction setting.

Thanks for the replies! 
Indeed lifting Z would resolve the issue. However, for this particular project, final finish is priority because it's a decor/display piece and lifting Z really does cause a lot of stringing and finish issues, even with "wipe" function on. In the end I managed to print these small parts by adding brim to help them sticking to the bed better and slowing the print speed waaaay down to 50%. That way the nozzle has time to melt its way thru the parts instead of knocking them down...

meteozond, 2026-09-23:
This is really needed :(