prusa3d/PrusaSlicer #9641: Please Add Seam Tuck (Similar to KISSlicer) (2.6.0)

limaciform, 2023-02-07:
**Is your feature request related to a problem? Please describe.**
I tried searching for 'KISSlicer', 'seams', and various other terms, but surprisingly didn't find this exactly (at least from the limited descriptions most requests had). The closest results were for continuous loop perimeters/multi-perimeter vase mode, but that is a different feature. There may very well be some sort of patent/IP protection that this feature has, but hopefully not. And if it's been raised before, then I can't apologize for raising it again, as it would be a really nice feature.

I never used KISSlicer back when it was popular, but I remember plenty of positivity around the quality of prints it produced. In those days, I found Slic3r to be the best for me, hence my love for PrusaSlicer/SuperSlicer. For a long, long time, KISSlicer seemed like it was a dead project. However, a couple of months ago a new version was released, so I decided to give it a try. It's still very far behind every other slicer as far as features go, but it still seems leaps and bounds ahead of other slicers when it comes to seams.

In PrusaSlicer, besides hiding seams in corners and not so visible areas (which PrusaSlicer does a good job of right now), there aren't many options for minimizing seams. Sometimes, you just can't hide a seam, but that seam can be detrimental to certain projects. For example, if you have a cylindrical pin that needs to insert or pivot within a hole (especially if it's a print-in-place), that seam can stop the parts fitting together, cause the pivot action to lock up, or if tolerances are too tight the pin may become one with the hole.

Here is a comparison of PrusaSlicer and KISSlicer, with all settings as close as possible. The only difference is the way seams are handled by each slicer.

External surface of a ring/cylinder:
[image]

Internal surface of a ring/hole:
[image]

**Describe the solution you'd like**
KISSlicer has a really nice way of dealing with seams, and the free version is available to download for any testing type demonstrations. 

Here is an explanation for a cylinder with 4 perimeters:
[image]

And in the case of a ring like the one demonstrated above (but much larger to allow 4 perimeters), KISSlicer deals with one wall/shell/set of perimeters first. So it starts with the outer wall of the ring, prints the outermost inner perimeter > external perimeter > inner perimeters outward in > the moves on to the inner wall of the ring and does the same, then adds infill.
[image]

**Describe how it would work**
- A simple tick box to enable 'seam tuck' or whatever feature name has a better sound.

- If one perimeter is chosen, this will not work/be disabled/tick box option greyed out.

- If 2 or more perimeters are chosen, this will work/be enabled/tick box option allowed.

- Print order will follow the same as KISSlicer, but the outermost inner perimeter will be treated with the same speed/acceleration etc. settings as the external perimeter (effectively printing 2 external perimeters).

- Likely will need a 'spacing' type input to tweak how steep the step out/step in angle is (i.e. the gap between the start/end of the outermost inner perimeter and the start/end of the external perimeter (in case its desirable to have an inset seam with zero bulge, or to try and tune it to be almost invisible). 

**Describe alternatives you've considered**
There is none I can think of. Coasting in Cura is very hit and miss, and can leave weak gaps. This just seems like the cleanest, easiest, and most reliable method.

gudvinr, 2023-02-07:
If you have tuned linear advance/pressure advance, these seams should not buldge.  
Coasting in cura uses same-ish principle as LA/PA but has way less control over extruder movement and will always be a subpar solution to proper LA/PA in firmware.

limaciform, 2023-02-07:
> If you have tuned linear advance/pressure advance, these seams should not buldge. Coasting in cura uses same-ish principle as LA/PA but has way less control over extruder movement and will always be a subpar solution to proper LA/PA in firmware.

If any of my printers _had_ linear/pressure advance, I would tune it.

Regardless, that's a _firmware_ thing, and this is a _slicer_ thing. Since _PrusaSlicer_ and _KISSlicer_ are both _slicers_, _firmware_ shouldn't be relied upon to make up for suboptimal _slicing_ methods. There's a broad range of _firmware_ flavours out there that all perform slightly differently depending on the manufacturer.

This demonstration shows that _KISSlicer_ handles seams better than _PrusaSlicer_, without relying on _firmware_. Besides acceleration, there's no _firmware_ affect on flow at all, just smarter gcode paths produced by the _slicer_.

So, if one _slicer_ solves a problem in the other _slicer_, purely by the better way it generates gcode paths to handle seams and walls, why should the better method not be used in PrusaSlicer?

gudvinr, 2023-02-07:
> If any of my printers had linear/pressure advance, I would tune it.

Cold hard truth is that *Prusa* Slicer made by developers from Prusa3D (company), who happen to make printers.  
And their printers do support LA.

So, when you ask them for features you should always keep this in mind. Business is business and if something does not benefit print quality on their printers, it probably would never be implemented.

> if one slicer solves a problem in the other slicer, purely by the better way it generates gcode paths to handle seams and walls, why should the better method not be used in PrusaSlicer?

I am not saying that your suggestion is bad or not beneficial in some ways, but only reason this could be considered is if it makes prints on prusa printers better. And if it isn't, it doesn't matter if this method performs better on other printers (which is your end goal basically).

HOLLY-DION, 2023-02-08:
The seams and bulges of the cylinders have really bothered me for too long

closed 2026-09-29 by prusaslicer-bot[bot]