prusa3d/PrusaSlicer #9642: Organic tree supports collide with models when printing multiple parts

Seba112PL, 2023-02-07:
### Description of the bug

When there is more than one part on the bed organic support can collide with a model.
The support material does not collide with the model it supports. It can, however, collide with the model next to it as well as with other support structures for other material.

[image]
[image]
[image]

### Project file & How to reproduce

place multiple object that needs support and enable organic support

### Checklist of files included above

- [ ] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.6 alpha 3

### Operating system

win 11

### Printer model

not related

neophyl, 2023-02-07:
This has always been the case with any type of supports in Prusa Slicer.  Its also been reported multiple times since 2.6 was released.  For example https://github.com/prusa3d/PrusaSlicer/issues/9614

Every object is sliced separately.  Its just more noticeable with organic supports as they extend outwards from the model more.  You have to merge the objects to be a single object and then the supports take notice of the other parts.  Please read https://github.com/prusa3d/PrusaSlicer/issues/9514

lukasmatena, 2023-02-07:
Closing as a duplicate of #316 (and others).

closed 2023-02-07 by lukasmatena