prusa3d/PrusaSlicer #9643: Organic supports not considering the size of of supported objects.

AmadeoMeure, 2023-02-07:
### Description of the bug

Im mostly printing miniatures so I get wierd edge cases a lot. I noticed one of my miniature's skirt only has a single support pillar under it until it connects with the main body. However, this object is far too large to only be supported by a single connection.

Proposed solutions
- Allow us to force the creation of a "connection point" for the supports.
         This would be my preferred option as computers can't always be trusted.
- increase the number of supports that can exist next to each other even if the slope of the object technically wouldn't need it.

[image]
skirt with single support showing the first layer with a connection to another support/model structure.

for this I used 0 overhang allowed and 100 tree support branch density and support everywhere.

[image]
Same problem on a far larger scale. It's only the base plate on the edge where it's touching the ground (which, due to it being a circle, is absolutely tiny) and the far rim. This is unprintable as is.,

### Project file & How to reproduce

[aliza combat alone.zip](https://github.com/prusa3d/PrusaSlicer/files/10674578/aliza.combat.alone.zip)

Reproduction is easy, just slice it with organic supports. I havnt found any settings that fix it yet so all of them should produce this error. To be specivic though,I used 0 overhang allowed and 100 tree support branch density and support everywhere.

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

Version 2.6.0-alpha3+win64

### Operating system

Windows 10

### Printer model

mini+

loskexos, 2023-02-08:
Ran into an issue with organic support that would match your description "not considering the size it supports".
Hope it is ok to add here in this issue?

In my case organic support ends with a tiny interface at a downwards pointing tip.
Print result was that it effectively failed there.

[image]

I'm now trying what happens with the whole cone-shaped part having paint-on-support.

[image]

Version 2.6.0-alpha3+win64 / Windows 11 / modified mk3s+

Moorviper, 2023-02-09:
Also on macos

 
[image]

[image]

DerGenaue, 2023-02-10:
You should use paint-on supports for such use-cases,
especially the new "Automatic Support Painter" that was introduced in this alpha version.
The release notes specifically recommend to use it with the new organic supports.

Other than that, this issue might still cause fewer supports to be generated:
https://github.com/prusa3d/PrusaSlicer/issues/9643

kubispe1, 2023-02-13:
@AmadeoMeure, thank you for the report. This issue is fixed probably with an overhang threshold fix.
So, in the upcoming release, it should be more stable. Closing this issue.

[image]
[image]

closed 2023-02-13 by kubispe1