prusa3d/PrusaSlicer #9645: No pad on branching support - SLA/Resin printer

Eloque, 2023-02-07:
### Description of the bug

When generating supporting with the branching feature activated, no support pad is generated. The supports connect directly to the build plate.

### Project file & How to reproduce

[Plain20-NoPad.zip](https://github.com/prusa3d/PrusaSlicer/files/10677001/Plain20-NoPad.zip)
[image]

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

PrusaSlicer-2.6.0-alpha3+win64-202302031521_signed

### Operating system

Windows 10

### Printer model

Photon Mono 4k

Eloque, 2023-02-20:
This seems to be solved in PrusaSlicer 2.6.0-alpha4

closed 2023-02-20 by Eloque

dpeet, 2023-03-14:
@Eloque I'm having the same issue on alpha5, but did not have it on alpha4.  It happens on both branching and regular support