prusa3d/PrusaSlicer #9646: Unable to Slice Object Close to build volume

WatchBenGo, 2023-02-07:
### Description of the bug

I am trying to slicer a larger file and take advantage of the new Organic supports.  I can slice this file just fine in 2.5.0.  However, when I load the model (with the same scale) into 2.6 is complains that the model is too large. 

Image from 2.5
[image]

Error in 2.6
[image]

As part of looking into this I also found a workaround that I can't explain.  I opened the model in 2.5, scaled it to the size I wanted, exported it from 2.5 as a STL, and once I did this I can import it into 2.6 and slice it without issue.  

(I've included both the scaled base model and the scaled and exported STL to assist with troubleshooting)

[Size Issue Files.zip](https://github.com/prusa3d/PrusaSlicer/files/10677156/Size.Issue.Files.zip)

### Project file & How to reproduce

See the two attached project files (one from 2.5 and one from 2.6) and try to slice the files. 

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

PrusaSlicer 2.6.0-alpha3

### Operating system

Windows 10 

### Printer model

Prusa i3 MK3S

neophyl, 2023-02-07:
This looks like a duplicate of https://github.com/prusa3d/PrusaSlicer/issues/9519

WatchBenGo, 2023-02-07:
@neophyl  - I believe you are correct!  I'll close this one out.

closed 2023-02-07 by WatchBenGo