prusa3d/PrusaSlicer #9649: New version, old problems. [Seams]

ramosglauco, 2023-02-08:
### Description of the bug

The quality of the seams generated by PrusaSlicer has always been "questionable", but since version 2.4 it has become terrible. I can't use the 2.5/2.6 versions because the seam algorithm works very very badly, especially on more complex polygonal models and also on models with cylindrical shapes. The paint option is also unresponsive, and often the seams are generated in a messy way. It is clear that there is great difficulty in placing the seams in the sharpest corners and also aligning them correctly. There also seems to be a preference for the algorithm to create wave patterns on the seams, which did not happen in previous versions (2.4 and below). It is extremely rare to be able to slice a model with an acceptably generated seam. For me, this is by far the worst disadvantage of the PS, because in my business the aesthetics of the models is a very important factor. Thank you and sorry for the bad english. This is not my native language.

**All the models below were generated with the ALIGNED option activated, without using seam painting. What was expected: the algorithm finds the sharpest corner in the model to place the seam.**

[image]

[image]

[image]

[image]

[image]

[image]

[image]

[image]

### Project file & How to reproduce

[3MF.zip](https://github.com/prusa3d/PrusaSlicer/files/10681376/3MF.zip)

### Checklist of files included above

- [ ] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.6-alpha3

### Operating system

Win11

### Printer model

Artillery, BIQU, Flsun

Snuff1eupagus, 2023-02-12:
I'm also having this problem, since version 2.5 the seams are awful, the seams in 2.4.2 were much better looking and could be controlled (placed easily) these seams are very difficult to work with, almost impossible to control. It seems like they go out of their way not to be where and how you want them. These seams are making my work look bad, people don't like them.

Can we please have the seams from 2.4 back? Pretty please, with sugar on top.

Operating system
Win 7 - 10 

Printer model
MK3

ramosglauco, 2023-02-12:
That should be a high-level priority, but apparently it makes no difference to them.

ramosglauco, 2023-02-12:
I believe this should be a high-level priority. Can you please help us with this issue? @bubnikv @lukasmatena

Snuff1eupagus, 2023-02-13:
Quick question? @Kubispe1 and @DragonFinlay who are you? I've been on GitHub not just for prusa, for several years and I don't recognize you. Are you affiliated with Prusa? Are you part of the dev. team? Whats going on? Who are we talking too? 

I have this nagging feeling that I'm wasting my time.

mix579, 2023-02-13:
I was just reminded of this issue today. Trying to slice a rectangular object with filleted sides, and the (aligned) seams down the edge look like the path of a drunken sailor. I was able to get better results by manually snapping a straight line with an unsupported method but this really needs to be addressed. I get the desire to add new cool stuff like text and cutting but let's not forget some of the basics.

RobinWts, 2023-02-15:
I agree with all of the above. Getting seams to work properly (again) should be a priority.

ramosglauco, 2023-02-15:
Please go back to the seams of the 2.4.x versions. They weren't perfect, but at least they were usable.

ramosglauco, 2023-03-03:
Hi, any updates on this issue?

Jan-Soustruznik, 2024-09-16:
closed as duplicate of https://github.com/prusa3d/PrusaSlicer/issues/8443

closed 2024-09-16 by Jan-Soustruznik