prusa3d/PrusaSlicer #9650: Access Violation Error

MadCatUSA, 2023-02-08:
### Description of the bug

Slicing a set of parts on the build plate. Added paint-on supports. Getting "Access Violation" error when slicing the parts. 
[image]

### Project file & How to reproduce

Can not save. File size limit is 25mb and file is 42mb.

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.6.0+Alpha 3+Win 64

### Operating system

Windows 11

### Printer model

Sovol SV06

kubispe1, 2023-02-08:
Hi, can you provide us with 3mf? Could help us to test and ensure, that Access Violation Error is fixed with some new commits?

kubispe1, 2023-02-08:
`Can not save. File size limit is 25mb and file is 42mb.`
For example, share it via an external cloud.

MadCatUSA, 2023-02-08:
I’m at work right now, but will try to email it when I get home this
evening.

voland696, 2023-02-08:
Do you have TIMELAPSE_TAKE_FRAME in the Before layer change G-Code ?

voland696, 2023-02-08:
Seems like if multi material painting used almost anything in Before layer change G-Code crushing slicer

MadCatUSA, 2023-02-08:
I don’t believe so, no, because I don’t do any Timelapse shots.

voland696, 2023-02-08:
Ok, in my case - I'm getting same "Access violation" error if anything besides G92 E0 in Before layer change G-Code.
For instance  TIMELAPSE_TAKE_FRAME. But if I rename TIMELAPSE_TAKE_FRAME to TESTTTT I have the same error. BUT! 
";TESTTTT" works, so workaround is to save gcode and replace ";TESTTTT" with TIMELAPSE_TAKE_FRAME

MadCatUSA, 2023-02-09:
[image: Zip Archive]
cad_bane_hat_HIGH [v1.0] [Eastman…
<https://drive.google.com/file/d/14ELxTzC4SvJyN_tynBIIJH3z5jl3ELC7/view?usp=drivesdk>

kubispe1, 2023-02-10:
Thanks I have it. We will check it.

kubispe1, 2023-02-10:
I can confirm that PrusaSlicer-2.6.0-alpha3+win64 produces Acess Violation Error.
Due to a new commits in the master. The issue is fixed until master PrusaSlicer-2.6.0-alpha3+33. So in the next release issue should not appear. 
I am closing this issue. Thanks

closed 2023-02-10 by kubispe1