prusa3d/PrusaSlicer #9651: Won't slice with multicolor / multiextruder Profile

dodasch, 2023-02-08:
### Description of the bug

Slicing Process stucks at 88% (Generating skirt and brim.) with PrusaSlicer-2.6.0-alpha3+win64-202302031521_signed
Slicing will work with "PrusaSlicer-2.5.0-rc2+win64-202208301346" with same Project file.

[image]

### Project file & How to reproduce

[Cap Logo.zip](https://github.com/prusa3d/PrusaSlicer/files/10684166/Cap.Logo.zip)

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

PrusaSlicer-2.6.0-alpha3+win64-202302031521_signed

### Operating system

Windows 10

### Printer model

Prusa i3Mk3s

lukasmatena, 2023-02-08:
Thanks, reproduced. We are looking into it.

bubnikv, 2023-02-17:
0202eec4b715eaa28a79f8bb82b0330c1efce8c9
    Fixed Clipper library (our own fork of it) when working with Z coordinate:
    The Eigen vector type compares all components, while the ClipperLib
    own IntPoint type compared x and y only.
    Fixed by overriding the ==/!= operators to compare just x and y components
    for Eigen types.

closed 2023-02-17 by bubnikv