prusa3d/PrusaSlicer #9655: Plater not visible on additional display

jafa74, 2023-02-08:
### Description of the bug

I can't see the plater on Mac iOS Ventura 13.2 when I'm using multiple display, on the laptop display I can normally but when I slide the window to an extended display de palter is blocked.

This error is present also when I open the software directly on the extended display

### Project file & How to reproduce

no report from the software

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.5.0+MacOS-x64

### Operating system

Ventura 13.2 (22D49)

### Printer model

Prusa MK3

jafa74, 2023-02-08:
https://user-images.githubusercontent.com/124689969/217620287-567e6fbc-edab-4136-9800-f4a544c1c962.mov

bubnikv, 2023-02-09:
How about any other 3D software, for example Cura or Fusion360?
Do they have the same issue?

pavelkestefran, 2023-05-12:
I have the same issue. On main display open with plater but when started on secondary display the plater view is blank. 
PrusaSlicer 2.5.2, MacOS Monterey 12.6.5

closed 2026-09-29 by prusaslicer-bot[bot]