prusa3d/PrusaSlicer #9656: 1 raft for print area

Savage79, 2023-02-08:
When printing multiple items, and using raft function, rafts will overlap if the items are to close.

**Describe the solution you'd like**
Have 1 raft, no overlaps for given print area regardless of items being printed. 

**Describe how it would work**
have an option to select raft type, for individual parts or print area.

**Describe alternatives you've considered**
currently use an alternate slicer (Cura), or space the items farther apart, longer print times, more material used.

**Additional context**
Add any other context or paint drawing about the feature request here.
Prusa style
[image]
[image]

Cura style
[image]
[image]

jmd73, 2023-02-09:
You can select all objects and use the merge command. Then all objects will be treated as one object and create a single raft.

rtyr, 2023-03-06:
Known issue. I am closing this issue as a duplicate of https://github.com/prusa3d/PrusaSlicer/issues/316

closed 2023-03-06 by rtyr