prusa3d/PrusaSlicer #9657: uncomfortable gizma, blocks the cut of fine details

Goodfeat, 2023-02-09:
**Is your feature request related to a problem? Please describe.**
When cut, the gizma obscures a thin object, and this does not make it possible to make a visual cut where it is needed.

**Describe the solution you'd like**
Move the central gizma some distance from the center of the cut

**Describe how it would work**
In older versions, the gizma was more conveniently located and did not block the part in the center

**Describe alternatives you've considered**
just move the gizma a distance

**Additional context**
old gizma
[image]

new gizma
[image]

solution to this problem
[image]

foreachthing, 2023-02-09:
The gizmo could also stay the same size, relative to the screen. That way it would not zoom with view.

Goodfeat, 2023-02-09:
In the current version of the gism, it scales as the screen zooms in and out.
On earlier versions of the slicer, a small offset of the gizma did not create any problems.

kubispe1, 2023-02-16:
Hi, this enhancement will be part of the next release. Closing this issue
SPE-1464
Commit: b40473be51fd1050fa4d7b2c3297230758085d46

closed 2023-02-16 by kubispe1