prusa3d/PrusaSlicer #9661: Bulge issue on thin wall only

NicolasGracia, 2023-02-09:
### Description of the bug

Hello everyone, 

I have bulge issue on thin wall and i try everything but always there. 
Same settings on S3D and no bulge. 
Why ? 
Apparently i'm not alone.

I would like understand this bug or problem.
Thank you everyone for your answers

### Project file & How to reproduce

[image]

### Checklist of files included above

- [ ] Project file
- [X] Screenshot

### Version of PrusaSlicer

Prusa slicer 2.6 alpha 3

### Operating system

Windows 11

### Printer model

Biqu B1

rtyr, 2023-02-09:
Please provide your 3MF project file. Are you using arachne slicing engine?

NicolasGracia, 2023-02-10:
Hi ! 
There 3MF File and i try with and without arachne engine. Same issue.
[Corps1.zip](https://github.com/prusa3d/PrusaSlicer/files/10704945/Corps1.zip)

NicolasGracia, 2023-02-14:
> Please provide your 3MF project file. Are you using arachne slicing engine?

Hi !
There 3MF File and i try with and without arachne engine. Same issue.
[Corps1.zip](https://github.com/prusa3d/PrusaSlicer/files/10729790/Corps1.zip)

rtyr, 2023-02-14:
thanks, I will try it

NicolasGracia, 2023-02-14:
> thanks, I will try it

Thanks a lot !

rtyr, 2023-02-16:
Provided 3MF is not PrusaSlicer project file. (File - Save project as).

NicolasGracia, 2023-02-26:
I found a solution. Thanks for your time :D !

I just put small perimeters and external perimeters at the same speed and VOILA. Bulge issue fix for me. 

I looked for the differences between my settings on S3D and prusaslicer and the only differences were this: There are not as many speed settings on S3D. The speed is almost fixed (only outside and inside speed but not "small perimeter or something")

rtyr, 2023-02-26:
Ok. You may also try pressure equalizer feature (it can gradually slow down the speed of the external perimeter to the same speed as external perimeter).

closed 2023-02-26 by rtyr