prusa3d/PrusaSlicer #9665: Crash at macOS when Orgnanic Support selected

jirka-malvin, 2023-02-09:
### Description of the bug

Alpha3 - When I choose organic support -> slice, application crash 

report from Mac included

### Project file & How to reproduce

choose organic support -> slice...
(with other supports it works)

-------------------------------------
Translated Report (Full Report Below)
-------------------------------------

Process:               PrusaSlicer [35741]
Path:                  /Applications/PrusaSlicer - testovaci.app/Contents/MacOS/PrusaSlicer
Identifier:            com.prusa3d.slic3r/
Version:               PrusaSlicer PrusaSlicer-2.6.0-alpha3+MacOS-x64-202302031521 (PrusaSlicer-2.6.0-alpha3+MacOS-x64-202302031521)
Code Type:             X86-64 (Native)
Parent Process:        launchd [1]
User ID:               501

Date/Time:             2023-02-09 15:42:47.4332 +0100
OS Version:            macOS 13.2 (22D49)
Report Version:        12
Bridge OS Version:     7.2 (20P3045)
Anonymous UUID:        C7226EA6-CE1F-30C0-28C1-8D428740106B

Sleep/Wake UUID:       7C7398A6-99A8-4A81-B372-65987E28094D

Time Awake Since Boot: 190000 seconds
Time Since Wake:       6188 seconds

System Integrity Protection: enabled

Crashed Thread:        10  slic3r_BgSlcPcs

Exception Type:        EXC_ARITHMETIC (SIGFPE)
Exception Codes:       0x0000000000000001, 0x0000000000000000

Termination Reason:    Namespace SIGNAL, Code 8 Floating point exception: 8
Terminating Process:   exc handler [35741]

Thread 0:: slic3r_main Dispatch queue: com.apple.main-thread
0   libsystem_kernel.dylib        	    0x7ff8090455c2 mach_msg2_trap + 10
1   libsystem_kernel.dylib        	    0x7ff809053604 mach_msg2_internal + 82
2   libsystem_kernel.dylib        	    0x7ff80904c635 mach_msg_overwrite + 723
3   libsystem_kernel.dylib        	    0x7ff8090458a8 mach_msg + 19
4   CoreFoundation                	    0x7ff80915fcbe __CFRunLoopServiceMachPort + 145
5   CoreFoundation                	    0x7ff80915e72a __CFRunLoopRun + 1360
6   CoreFoundation                	    0x7ff80915db60 CFRunLoopRunSpecific + 560
7   HIToolbox                     	    0x7ff812aab766 RunCurrentEventLoopInMode + 292
8   HIToolbox                     	    0x7ff812aab576 ReceiveNextEventCommon + 679
9   HIToolbox                     	    0x7ff812aab2b3 _BlockUntilNextEventMatchingListInModeWithFilter + 70
10  AppKit                        	    0x7ff80c2ae293 _DPSNextEvent + 909
11  AppKit                        	    0x7ff80c2ad114 -[NSApplication(NSEvent) _nextEventMatchingEventMask:untilDate:inMode:dequeue:] + 1219
12  AppKit                        	    0x7ff80c29f757 -[NSApplication run] + 586
13  PrusaSlicer                   	       0x103bf4a7a wxGUIEventLoop::OSXDoRun() + 186
14  PrusaSlicer                   	       0x103cdfd1d wxCFEventLoop::DoRun() + 29
15  PrusaSlicer                   	       0x103c56755 wxEventLoopBase::Run() + 85
16  PrusaSlicer                   	       0x103c2e4a4 wxAppConsoleBase::MainLoop() + 116
17  PrusaSlicer                   	       0x103bb47ca wxApp::OnRun() + 26
18  PrusaSlicer                   	       0x103c87df9 wxEntry(int&, wchar_t**) + 73
19  PrusaSlicer                   	       0x101c9d9bc Slic3r::GUI::GUI_Run(Slic3r::GUI::GUI_InitParams&) + 236
20  PrusaSlicer                   	       0x1010a6a77 Slic3r::CLI::run(int, char**) + 21159
21  PrusaSlicer                   	       0x1010ab9f0 main + 352
22  dyld                          	    0x7ff808d51310 start + 2432

Thread 1:: com.apple.NSEventThread
[... 365 lines cut ...]
USB Device: Touch Bar Display
USB Device: Apple Internal Keyboard / Trackpad
USB Device: Headset
USB Device: Ambient Light Sensor
USB Device: FaceTime HD Camera (Built-in)
USB Device: Apple T2 Controller
Thunderbolt Bus: MacBook Pro, Apple Inc., 47.5
Thunderbolt Bus: MacBook Pro, Apple Inc., 47.5

### Checklist of files included above

- [X] Project file
- [ ] Screenshot

### Version of PrusaSlicer

in report

### Operating system

MacOS 13.2 Ventura

### Printer model

my version of 3D printer - not rellevant

bubnikv, 2023-02-09:
Division by zero in Organic supports
FFFTreeSupport::generate_support_areas()

kubispe1, 2023-02-17:
Potentially duplicate: #9555

lukasmatena, 2023-02-18:
Should be fixed in [2.6.0-alpha4](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.6.0-alpha4). Please retest and let us know if problems persist. Closing.

jirka-malvin, 2023-02-23:
Lukasi, 
problem persist. 
Jirka

P.S. Kdyz budete chtit, muzu za Vami do Holesovic dorazit a nazorne ukazat.

-------------------------------------
Translated Report (Full Report Below)
-------------------------------------

Process:               PrusaSlicer [83766]
Path:                  /Applications/PrusaSlicer.app/Contents/MacOS/PrusaSlicer
Identifier:            com.prusa3d.slic3r/
Version:               PrusaSlicer PrusaSlicer-2.6.0-alpha4+MacOS-x64-202302180021 (PrusaSlicer-2.6.0-alpha4+MacOS-x64-202302180021)
Code Type:             X86-64 (Native)
Parent Process:        launchd [1]
User ID:               501

Date/Time:             2023-02-23 10:50:55.8317 +0100
OS Version:            macOS 13.2 (22D49)
Report Version:        12
Bridge OS Version:     7.2 (20P3045)
Anonymous UUID:        C7226EA6-CE1F-30C0-28C1-8D428740106B

Sleep/Wake UUID:       09ABE280-DC9F-4AC3-9E17-9A50F4C1CCEC

Time Awake Since Boot: 320000 seconds
Time Since Wake:       4290 seconds

System Integrity Protection: enabled

Crashed Thread:        15  slic3r_BgSlcPcs

Exception Type:        EXC_ARITHMETIC (SIGFPE)
Exception Codes:       0x0000000000000001, 0x0000000000000000

Termination Reason:    Namespace SIGNAL, Code 8 Floating point exception: 8
Terminating Process:   exc handler [83766]

Thread 0:: slic3r_main Dispatch queue: com.apple.main-thread
0   libsystem_kernel.dylib        	    0x7ff8185d95c2 mach_msg2_trap + 10
1   libsystem_kernel.dylib        	    0x7ff8185e7604 mach_msg2_internal + 82
2   libsystem_kernel.dylib        	    0x7ff8185e0635 mach_msg_overwrite + 723
3   libsystem_kernel.dylib        	    0x7ff8185d98a8 mach_msg + 19
4   CoreFoundation                	    0x7ff8186f3cbe __CFRunLoopServiceMachPort + 145
5   CoreFoundation                	    0x7ff8186f272a __CFRunLoopRun + 1360
6   CoreFoundation                	    0x7ff8186f1b60 CFRunLoopRunSpecific + 560
7   HIToolbox                     	    0x7ff82203f766 RunCurrentEventLoopInMode + 292
8   HIToolbox                     	    0x7ff82203f576 ReceiveNextEventCommon + 679
9   HIToolbox                     	    0x7ff82203f2b3 _BlockUntilNextEventMatchingListInModeWithFilter + 70
10  AppKit                        	    0x7ff81b842293 _DPSNextEvent + 909
11  AppKit                        	    0x7ff81b841114 -[NSApplication(NSEvent) _nextEventMatchingEventMask:untilDate:inMode:dequeue:] + 1219
12  AppKit                        	    0x7ff81b833757 -[NSApplication run] + 586
13  PrusaSlicer                   	       0x107fdaffa wxGUIEventLoop::OSXDoRun() + 186
14  PrusaSlicer                   	       0x1080c629d wxCFEventLoop::DoRun() + 29
15  PrusaSlicer                   	       0x10803ccd5 wxEventLoopBase::Run() + 85
16  PrusaSlicer                   	       0x108014a24 wxAppConsoleBase::MainLoop() + 116
17  PrusaSlicer                   	       0x107f9ad4a wxApp::OnRun() + 26
18  PrusaSlicer                   	       0x10806e379 wxEntry(int&, wchar_t**) + 73
19  PrusaSlicer                   	       0x106085938 Slic3r::GUI::GUI_Run(Slic3r::GUI::GUI_InitParams&) + 248
20  PrusaSlicer                   	       0x10548fcb7 Slic3r::CLI::run(int, char**) + 21159
21  PrusaSlicer                   	       0x105494c30 main + 352
22  dyld                          	    0x7ff8182e5310 start + 2432

Thread 1:: com.apple.NSEventThread
0   libsystem_kernel.dylib        	    0x7ff8185d95c2 mach_msg2_trap + 10
1   libsystem_kernel.dylib        	    0x7ff8185e7604 mach_msg2_internal + 82
2   libsystem_kernel.dylib        	    0x7ff8185e0635 mach_msg_overwrite + 723
3   libsystem_kernel.dylib        	    0x7ff8185d98a8 mach_msg + 19
4   CoreFoundation                	    0x7ff8186f3cbe __CFRunLoopServiceMachPort + 145
[... 353 lines cut ...]
AirPort: spairport_wireless_card_type_wifi (0x14E4, 0x7BF), wl0: Dec  1 2022 17:15:41 version 9.30.489.6.32.5.85 FWID 01-c30a5043
Bluetooth: Version (null), 0 services, 0 devices, 0 incoming serial ports
Network Service: USB3.0 5K Graphic Docking, Ethernet, en5
Network Service: Wi-Fi, AirPort, en0
USB Device: USB3.0 Hub
USB Device: USB5734
USB Device: USB3.1 Hub
USB Device: USB3.0 5K Graphic Docking
USB Device: USB31Bus
USB Device: USB2.0 Hub
USB Device: Generic Billboard Device
USB Device: Dell KB216 Wired Keyboard
USB Device: USB2734
USB Device: Hub Controller
USB Device: USB2.1 Hub
USB Device: T2Bus
USB Device: Touch Bar Backlight
USB Device: Touch Bar Display
USB Device: Apple Internal Keyboard / Trackpad
USB Device: Headset
USB Device: Ambient Light Sensor
USB Device: FaceTime HD Camera (Built-in)
USB Device: Apple T2 Controller
Thunderbolt Bus: MacBook Pro, Apple Inc., 47.5
Thunderbolt Bus: MacBook Pro, Apple Inc., 47.5

lukasmatena, 2023-02-23:
@jirka-malvin Thanks for the offer. At this point, it might be enough to provide a project file (3mf) that suffers from this problem. Could you please post it here?

bubnikv, 2023-03-03:
@jirka-malvin Thanks for the offer. At this point, it might be enough to provide a project file (3mf) that suffers from this problem. Could you please post it here?

jirka-malvin, 2023-03-03:
[medieval_Revised.3mf.zip](https://github.com/prusa3d/PrusaSlicer/files/10886918/medieval_Revised.3mf.zip)

enclosed ... Jiri

lukasmatena, 2023-03-06:
This crash is fixed in 2.6.0-alpha4. Closing.

lukasmatena, 2023-03-06:
We are unable to reproduce the problem in 2.6.0-alpha4.

jirka-malvin, 2023-03-06:
will it help you if i come with laptop and show you personally?
Jiri

jirka-malvin, 2023-03-06:
> @jirka-malvin <https://github.com/jirka-malvin> I just tried to recompile 2.6.0-alpha4 in Debug mode and found a failling assert, which may lead us to the cause of the problem. Could you please try to disable Print Settings -> Layer and Perimeters -> Avoid crossing perimeters and let us know if the crash still occurs? Thanks.
>
> By the way, your model is hideously broken. I would assume that fixing it will make things better. That's of course not an excuse, PrusaSlicer should not crash on it. I just wouldn't be surprised if the slicing was still wrong even if the crash is fixed.

still persist with crossing perimetrs disabled ..

kubispe1, 2023-04-14:
I probably detect the issue. I used @jirka-malvin 's model and turn on **Organic supports**. Then I can reproduce crash for my **Mac and even for Win**. Issue is in 2.6.0a4, 2.6.0a6, 2.6.0a6+73(actual master)
- My feeling is that model has quite a lot of open edges (I am not able to fix it in Netfabb) with ****combination support_material_treshold** = 90°**
- I think, deeper investigation is required. Can I ask you @lukasmatena to check it?
**3mf: [medieval_Revised_Fail.zip](https://github.com/prusa3d/PrusaSlicer/files/11232368/medieval_Revised_Fail.zip)**

lukasmatena, 2023-04-14:
@kubispe1 Thanks, this helped. It seems related to #10158.

kubispe1, 2023-04-14:
Ok fine, issue is selected for development: 
**SPE-1658**

kubispe1, 2023-05-17:
"This crash should be fixed with PS 2.6.0-beta1. If any issues persist, please let us know. Thank you.

closed 2023-05-17 by kubispe1