prusa3d/PrusaSlicer #9666: error

westloma, 2023-02-09:
### Description of the bug

error prusa-sliser

### Project file & How to reproduce

[image]

### Checklist of files included above

- [ ] Project file
- [X] Screenshot

### Version of PrusaSlicer

Version: 2.6.0-alpha2+win64 Build: PrusaSlicer-2.6.0-alpha2+win64-202301311700

### Operating system

Operating System: Windows System Architecture: 64 bit Windows Version: Windows 10 (сборка 19044), 64-бит редакция

### Printer model

ender 3 neo

lukasmatena, 2023-02-09:
Please, retest the problem in [2.6.0-alpha3](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.6.0-alpha3), several issues that might have caused it are fixed there.
If the problem persists in alpha3, we will need a project file so we can reproduce and fix it. Thanks.

westloma, 2023-02-09:
ошибка не исчезла
[labyrinthGiftBox_bottom.zip](https://github.com/prusa3d/PrusaSlicer/files/10698791/labyrinthGiftBox_bottom.zip)
использовал версию которую вы предоставили

Pilgrim182551, 2023-02-10:
the same issue with 2.6.0-alpha3

lukasmatena, 2023-02-10:
@westloma @Pilgrim182551 Guys, your STLs are nice, but I need a 3MF, which also contains your configuration. Set everything the way it would crash, the `File->Save Project` and send that file here.

westloma, 2023-02-10:
[labyrinthGiftBox_bottom.zip](https://github.com/prusa3d/PrusaSlicer/files/10707467/labyrinthGiftBox_bottom.zip)

Pilgrim182551, 2023-02-10:
updated: 

[set.zip](https://github.com/prusa3d/PrusaSlicer/files/10707633/set.zip)

lukasmatena, 2023-02-10:
@westloma Thanks. I just checked and found out that it is a duplicate of #9557, which was already fixed for 2.6.0-alpha4 (not released yet).

@Pilgrim182551 I will check later whether your issue is the same or not.

lukasmatena, 2023-02-13:
@Pilgrim182551 I just checked that your issue disappears when you switch to "Grid" supports and that it seems to be fixed for 2.6.0-alpha4. Indeed likely a duplicate of #9557. Closing.

closed 2023-02-13 by lukasmatena