prusa3d/PrusaSlicer #9670: Supports in cut connectors

dwbmb, 2023-02-09:
### Description of the bug

When using new "cut connectors" feature along with painting "Block supports" area in the same area where connectors are supposed to be, supports are still being generated inside these connector holes.

### Project file & How to reproduce

Create cut with connectors
paint "block supports" on upper purt of the cut model with holes for connectors
Have supporst generated

Voila, supports are present in blocked area too

[image]
[image]

### Checklist of files included above

- [ ] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.6.0-alpha3

### Operating system

Windows 11

### Printer model

Mini+

bubnikv, 2023-02-10:
Please provide a 3MF.
BTW there was a bug that the automatic support generator did not suppress
supports below bridges, see Issues · prusa3d/PrusaSlicer (github.com)
<https://github.com/prusa3d/PrusaSlicer/issues/9493>

dwbmb, 2023-02-11:
.3mf project added. Suppressing supports below bridges did not work too, you are right, but here I wanted to point out the supports are being generated on painted "block supports" areas too.

dwbmb, 2023-02-16:
There is "awaiting response" label here. Is there anything more I should deliver?

64bitcube, 2023-08-12:
I'm experiencing the same 'bug'.  I assume it's an oversight/just a complication of the multiple systems, but it really would be great if there was a solution.  Especially since I'm using the cut/connector method entirely to get a flat build area on my model, I don't want the support interface to interfere with the fit of the 'dowels'.  

Two solutions from a UI standpoint I see:
A) cut connectors within an area of 'supports blocked' should carry that property to the new interior surface, so no supports are generated.
B) Cut connectors could have a right click/'add modifier" menu that includes an on/off toggle for support generation.  If it's checked appropriately, the surface from the cut connector will refuse any support generation.

C-Glick, 2023-12-20:
I ran into this same issue. As a work around I was able to use a cube support blocker around the hole instead of painting on support blockers.

Irav, 2024-05-26:
Has this been updated? -- I got a large part that is making me waste more than 100g in unnecessary supports for the connectors.

Jan-Soustruznik, 2024-09-17:
Hello, @dwbmb, @Irav , Yes, issue should be fixed by commit https://github.com/prusa3d/PrusaSlicer/commit/f1977b07be304076be720b058757bc4d36158e83 mention there https://github.com/prusa3d/PrusaSlicer/issues/9493#issuecomment-1424411677  Now the support blocker or Dont support bridges in support matterial setting now works. I closed this issue as solved. You can check in latest release of 2.8.1-rc1(https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.8.1-rc1)

closed 2024-09-17 by Jan-Soustruznik