prusa3d/PrusaSlicer #9673: Slicer 2.6.0 alpha 3 Soluble support interface

cesaremonti, 2023-02-10:
### Description of the bug

When I sert the soluble support interface (doesn't matter if organic or linear supports) the slicer mix the layer of the base filament and the soluble
[image]

### Project file & How to reproduce

[Lhea 2023 Organic Tree supports.zip](https://github.com/prusa3d/PrusaSlicer/files/10708115/Lhea.2023.Organic.Tree.supports.zip)

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.6.0 alpha 3

### Operating system

windows 10 64

### Printer model

MK3S +MMU2S

vcapra1, 2023-02-16:
I'm seeing the same thing.  To add some info, the "feature type" is correct:
[image]

But the tool changes are wrong:
[image]

Almost looks like it's not correctly switching back to the non-interface tool when it's done printing the interface

vcapra1, 2023-02-17:
Narrowed it down to commit https://github.com/prusa3d/PrusaSlicer/commit/8858651bf.  It's broken on that commit but on https://github.com/prusa3d/PrusaSlicer/commit/6e653d90703457c8a0e08f421454b1ba6081c78d it works correctly.

closed 2026-09-29 by prusaslicer-bot[bot]