prusa3d/PrusaSlicer #9675: Timeout Error when uploading to Octoprint 1.8.6

jtleniger, 2023-02-10:
### Description of the bug

The first 1-3 times I click the button to send & print to OctoPrint version 1.8.6, I receive the following timeout error:

```
Error uploading to print host:
Timeout was reached:
Connection timed out after 10007 milliseconds
[Error 28]
```

I don't see any additional log messages or a verbose log anywhere so I'm not sure how to isolate the issue to OctoPrint or PrusaSlicer.

### Project file & How to reproduce

[Body1.zip](https://github.com/prusa3d/PrusaSlicer/files/10710806/Body1.zip)

Note: this happens on any project, no matter the file.

[image]

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.5.0+win64

### Operating system

Windows 10

### Printer model

Prusa i3 MK3S

drpurr1975, 2023-03-09:
I have exactly the same issue for first try to upload

tontonlex, 2023-05-13:
I have similar problem, when "download and print" dont work sometimes "download" work, But today anything work,
I dont know is it's linked, but it seems to happen more often when prusaslicer load is "long" and seem to stuck on "loading setup"
Octoprint v 1.8.7 and prusaslicer 2.5.2

jtleniger, 2023-10-01:
This issue appears to be resolved.

closed 2023-10-01 by jtleniger