prusa3d/PrusaSlicer #9679: Access Violation when slicing file (2.6.0-alpha3)

jsmith59, 2023-02-11:
### Description of the bug

Encountered error when slicing file in PrusaSlicer 2.6.0-alpha3. Fatal error: Access Violation. Slices without error in version 2.5.0.
[image]

### Project file & How to reproduce

Load file in PrusaSlicer and press Slice now.
[cat bathroom sign MK3S.zip](https://github.com/prusa3d/PrusaSlicer/files/10712289/cat.bathroom.sign.MK3S.zip)

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

PrusaSlicer 2.6.0-alpha3+win64

### Operating system

Windows 11

### Printer model

MK3S+ with MMU2S

kubispe1, 2023-02-13:
Hi, I can reproduce the issue. Thanks

lukasmatena, 2023-02-13:
Still crashes in `db77331` (2.6.0-alpha3+42). The crash happens in `process_layer_single_object` on `const LayerSlice &lslice = layer->lslices_ex[idx];` (the vector has a single element, idx is 1).

lukasmatena, 2023-02-15:
SPE-1474

bubnikv, 2023-02-16:
You had "clip multi-part objects" disabled.
Did you have a reason for that?

jsmith59, 2023-02-16:
I didn't realize it was disabled. I thought I was using a default profile.
I am currently out of town so I can't try slicing the model with the
feature enabled.

bubnikv, 2023-02-16:
Fixing with edd6c7c9d95aa2f46abe293c9bc89d619ec2491a and f78a2c0bf662af807bacc938b005279448a1aab9

The first commit removes the obscure "clip_multipart_objects" option, which caused more  harm than good and it is not\compatible with the new layer island Z graph.

The second commit improves robustness of the Z graph.

closed 2023-02-16 by bubnikv