prusa3d/PrusaSlicer #9684: Option "Thick bridges" makes bridges thin.

Fubaxiusz, 2023-02-11:
### Description of the bug

Option "Thick bridges" in the in the "Layers and Perimeters" section under "Quality (slower slicing)" makes bridges actually thin, not thick, by introducing gaps between lines. It should be called "Thin bridges".

### Project file & How to reproduce

[Nalewacz czekolady kornel.zip](https://github.com/prusa3d/PrusaSlicer/files/10714895/Nalewacz.czekolady.kornel.zip)
Look at layer number 25.
Select or deselect option "Thick bridges" in the "Layers and Perimeters" section under "Quality (slower slicing)".
See that the name of the setting does the opposite than it says,

[image]
*Normal bridges which are actually thick.*

[image]
*"Thick bridges" which are actually thin.*

[image]

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

Version 2.5.0+win64

### Operating system

Windows 10

### Printer model

Creality CR10

neophyl, 2023-02-11:
Try looking at the preview from the side. Thick refers to the height, as in they are assumed to be printed at the nozzle diameter thick.  So the extrusion is assumed to be a circular cross section equal to the nozzle diameter. 
Thin bridges changes that assumption to that the lines are your configured layer height.

rtyr, 2023-02-12:
Expected behavior. See https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.4.0-alpha1 for more info about this parameter.

closed 2023-02-12 by rtyr