prusa3d/PrusaSlicer #9686: missing geeetech A10-original printer profile

goofyseeker311, 2023-02-12:
missing profile: you have profiles A10-pro, A10M and A10T but not the original A10 3d printer from geeetech.

rtyr, 2023-02-12:
3rd party profiles are usually provided and maintained by the community, Geeetech bundle was provided by @alexwoo1900 and it is perfectly possible it does not contain every existing Geeetech model.

As for the other thing you mentioned, I am afraid I do not understand what exactly do you mean from the description. Please open separate issue with more information (PS version, operating system, steps to reproduce, ...) and mainly 3MF project file.

goofyseeker311, 2023-02-12:
Its just a problem I found while first time using the program on
win11pro/5800x/rtx3080/128gb/samsung990pro computer.

Sorry if you cant understand my description. After slicing the model
vanishes from the build plate even if its saved there.

rtyr, 2023-02-13:
It can be caused by printer profile. Please provide 3mf project file (File - Save project as.., then archive to zip).

goofyseeker311, 2023-02-13:
No, its not public or given to you.

rtyr, 2023-02-13:
So it does happen only with one particular model?

rtyr, 2023-02-13:
What printer profile are you using excatly?

goofyseeker311, 2023-02-13:
Its the only and first model that I used with prusaslicer recent with
aA10pro profile. I did not try it again, only reloaded the project 3mf file
after program restart that was saved when it did not show on the print bed.

rtyr, 2023-02-14:
I was not able to reproduce it.

goofyseeker311, 2023-02-16:
It might be related to absolutely first run-install-use of the prusaslicer, it could slice another model without having the issue.

additional usability notes: vertically stacking multiple (tree-supported, only one attach point) objects is kinda hard, easier if:
a) automatic vertical stacking arrangement of the objects, not only horizontal object automatic arrangement
b) copy-paste would paste the object to same location, manually vertically positioning the ojects
c) proper centering/object positioning tools

rtyr, 2023-02-16:
OK, I am closing this issue for now.

> additional usability notes

It is better to open separate issue for each topic, otherwise it is very hard to maintain it for us.

closed 2023-02-16 by rtyr