prusa3d/PrusaSlicer #9687: OpenSCAD Customizer Integration

DeadlyDad, 2023-02-12:
**Is your feature request related to a problem? Please describe.**
While OpenSCAD offers an almost unlimited to tune a model as deeply as one might wish, actually using it can be daunting to someone who isn't comfortable with code.  The Customizer feature bypasses this to a certain extent by allowing users to tweak a model to their liking by just adjusting a few parameters.  Unfortunately, actually *using* OpenSCAD to create--or even just tweak--a customized model can be... awkward for most users.

**Describe the solution you'd like**
I'd like to see OpenSCAD script--and, specifically, Customizer--support added to .3mf files and PrusaSlicer.  

**Describe how it would work**
When a .3mf file that contains an OpenSCAD script is opened up, an option is given to allow users to use Customizer--as used to much success with Thingiverse--to adjust the model's parameters just before printing.  (I wouldn't bother with a full editor window; if someone wants to tweak the code itself, they can just install OpenSCAD.)  This would **drastically** expand the usability of PrusaSlicer, as users could create--for example--custom pulleys/gears/boxes/etc. without needing to know any code or use any other software.

Internally, beyond incorporating the rendering engine, an 'OpenSCAD' folder would be added to the .3mf files structure, with an option to save user-generated Customizer profiles to it, so users can quickly reprint a custom model without needing to save another full STL.  Even better, scripts can import STLs and alter them as needed.  (As a bonus, model components can already have a set colour.)  Scripting upgrades could both take the form of new commands and special comment formats--such as are used by the Customizer--which are, essentially, 'compiler directives'.  

**Describe alternatives you've considered**
I can't think of *any* really practical alternatives.  OpenSCAD is already a well-established app with [a **LOT** of parametric models already available](https://www.thingiverse.com/?page=1&per_page=20&sort=popular&type=things&q=&customizable=1), and tweaking existing scripts to work with Customizer is [*ridiculously* easy](https://customizer.makerbot.com/docs).

Does that sound reasonable?

kubispe1, 2023-02-12:
Ok, thanks for FR.

EternityForest, 2023-05-24:
I love it!  I think you should be able to embed STLs that the customizer can access, along with font files. That way, it can be used to allow for minor customization of things not made in OpenSCAD, which isn't exactly the nicest way to do a complex model.

A major use case I can think of would be adding labels to things.  It would also give you a way to bundle stuff together in a single file for the shape gallery, using the drop-down feature to select among multiple included STLs.

I would almost go so far as to say in-app editing *is* worth it, to make it easier to add the parametric features in the first place, but it's not necessary.

As a separate but very related request, adding support for displaying a README.md file would be very useful for parametric models, to be able to explain what the parameters do in more detail.

DeadlyDad, 2023-05-25:
FWIW, OpenSCAD can [import](https://en.wikibooks.org/wiki/OpenSCAD_User_Manual/Importing_Geometry#import) various 2d and 3d formats for processing, so yes, you could import a (static) complex model, cut out/add a space for a label, add in text/etc., then export the customized version.

DeadlyDad, 2026-09-05:
None of the devs have really responded to this.  Input?