prusa3d/PrusaSlicer #9689: Please add Sovol SV01 Pro on the list as well

ricoshe81x, 2023-02-12:
Please add Sovol SV01 Pro to the model list. There is the Sovol SV01 but it would be great if the list gets completed.

gudvinr, 2023-02-12:
3rd party printers are not supported by prusa but rather added by community. You can send PR to [prusa3d/PrusaSlicer-settings](https://github.com/prusa3d/PrusaSlicer-settings) to add configuration for this printer.

kubispe1, 2023-02-12:
Profiles for 3rd party printers are created by the community.

3rd party profiles are best provided as pull requests at
https://github.com/prusa3d/PrusaSlicer-settings
We have a contributor guide.
https://github.com/prusa3d/PrusaSlicer/wiki/Vendor-bundles-and-updating-process

Thanks, I am Closing the issue.

closed 2023-02-12 by kubispe1