prusa3d/PrusaSlicer #9694: Acceleration control (Advanced) > Default - adds M204 to g-code

cosworth99, 2023-02-12:
### Description of the bug

I have Machine Limits set to ignore and it still injects an M204 into the g-code. It populates the number from Acceleration Control (Advanced) "Default" (500mm/sec). The default should be to NOT alter acceleration, with a zero in place of the 500. My printer is set to 3000mm/sec. Running this g-code sets the firmware acceleration to 500.

This field is still adding an "M204 500" with ignore set. It's an odd choice to force acceleration on a printer by default. My other slicers do not do this. This shouldn't have to be disabled, it should have to be enabled. Setting Acceleration control (Advanced) to 0mm/sec and saving the printer profile fixes the issue. 

Since it is a machine limit, it should fall under the ignore option. This is half feature request half bug report.

Acceleration control (Advanced) > Default  - should be zero by default. Thereby disabling it. 

### Project file & How to reproduce

To reproduce - Set "Machine Limits" to >General>Ignore.  Slicer populates g-code with the number 500 from Print Settings>Acceleration Control (Advanced)>Default. Slice and export. Open g-code with text editor and search M204.

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

Version 2.6.0-alpha3+MacOS-x64

### Operating system

Mac OS Monterey 12.6.3

### Printer model

Ender 5 Plus with SKR Mini E3 V2 and Marlin 2.1.2

rtyr, 2023-02-12:
I don´t see any problem here. Machine limits and acceleration control are two different things. If you set "ignore" or "use for time estimate", PS will not emit commands related to the FW limits at the start of the g-code. It is mainly about M201 when it comes to the acceleration, which is max acc limit (unlike M204).

> Acceleration control (Advanced) > Default - should be zero by default.

Default value is indeed 0, so it does not emit any M204 command by default. If you see some other value there, for example 500, than that is the value which was set intentionally by the author of the profile you are using (for example bundled Ender profile). I believe Creality used 500mms max acceleration limit in its stock FW, at least on early Enders.

closed 2023-02-12 by rtyr

cosworth99, 2023-02-12:
The issue is that it is a machine limit. If you choose to “ignore” - it should not write it to g-code. 

Going in a tab workflow from left to right, the machine limits should negate previous machine limits if you choose to ignore. 

It’s not a bug, it’s a UX issue. It’s a weak implementation. It alters the printer eeprom for subsequent prints when you have previously told it not to in your final tab - going in a left to right process. Plater>Print Settings> Filament>Printer Settings.

rtyr, 2023-02-13:
I don't think so. Acceleration control lets you change acceleration per feature and per print profile (for example you may want different acc for 50um and for 200um, for infill and for perimeters). Machine limits are about max. limits for printer in general. It is used mainly for calculating estimate time and you can choose whether you want those limits emitted to the gcode (override fw limits) or not. It doesn't matter if you send M204 S10000 if your FW limit is 500 (so 500 will be used).

Again, acceleration control is disabled by default. So don't use it if you don't want it.