prusa3d/PrusaSlicer #9695: Physical printer: Print Host Type keeps changing

tonerod, 2023-02-12:
### Description of the bug

I'm trying to setup Prusa slicer to print directly to my printer (Ender 3 S1 Pro running Klipper).  Everyting goes well in the setup, including a "Test for connection", but when I click "OK" which closes the "physical printer" window, and I open that window again, all my settings are still there, but the "Host Type" keeps switching from what I seclected (Octoprint) to FlashAir.  I can't get it to stay on "Octoprint"  Just as a test, I tried seclecting each of the other "Host Type" in the dropdown box, and saving.  Each one of them remain as I selected them, except "Octoprint"

Note. The process works fine when I set up my Prusa i3 mk3

### Project file & How to reproduce

1. Edit "physical printer" under printer setting
2. Fill out all the required fields, (seclect "Octoprint" as host type).
3. Press "Test" to check for connection... receive message "Connection to Octoprint works correctly"
4. Hit "OK" and window closes
5. Open window again and "FlashAir" is now in the "Host Type" field.
6. Can not get "Octoprint" to stay as "Host Type" 

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

Version 2.6.0-alpha3+win64

### Operating system

Windows 10

### Printer model

Ender 3 S1 Pro running Klipper

kocikdav, 2023-02-14:
Hello, thank you for reporting.
I can confirm, this bug is happening at my side too. We will look at it. 
I also checked and after point 4 - PS should be entering code for Octoprint upload. So uploading should work (until you re-enter the dialog.) Anyway it needs to be fixed.

jokesters, 2023-02-16:
I also have this bug on linux mx

kubispe1, 2023-02-16:
Thanks for your issue, the fix will be part of the next release.
commit: 2125996837c30c0fb51971528cb09265f2ee6517
SPE-1477

closed 2023-02-16 by kubispe1