prusa3d/PrusaSlicer #9699: Top Contact Z distance ignored when using Organic support material

dp250f, 2023-02-12:
### Description of the bug

When Organic support is enabled, top contact Z distance is always equal to the layer height. It acts like "Synchronize with object layers" is enabled when it is not. It allows Top contact distance to be set as expected but you can see in the preview it is just using layer height.

### Project file & How to reproduce

Set layer height to 0.25
Enable Grid support
Set top contact Z distance to 0.1
Observe small gap between top interface layer and supported layer (visible in preview)
Change to Organic support
Observe layer-height gap between top interface layer and supported layer (visible in preview)
[Transmission top.zip](https://github.com/prusa3d/PrusaSlicer/files/10717461/Transmission.top.zip)
[image]
[image]

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.6.0-alpha-3+win64

### Operating system

Windows 11 22H2

### Printer model

Prusa i3 MK3S (SKR 1.4 turbo, Marlin 2.1.2)

rtyr, 2023-02-13:
Expected behavior mentionied in Alpha2 release notes.

[image]

So in this case, you can use 0/0.25/0.50/...

thunderpipesusa, 2023-02-15:
When I use .20 layer height, and I have .2 detachable selected for top bottom interface… I get a .2 gap between the organic and the model.  Shows in ps and prints same way.  Organics never touch the model.

rtyr, 2023-02-15:
> When I use .20 layer height, and I have .2 detachable selected for top bottom interface… I get a .2 gap between the organic and the model. Shows in ps and prints same way. Organics never touch the model.

That is correct, nothing wrong about that.

dp250f, 2023-02-18:
Oh. That's too bad. The ability to fine-tune Z-contact distance is one of the reasons I prefer PrusaSlicer. Hopefully this functionality can be added to Organic supports in the future.

Warsenius, 2023-03-10:
im also facing this challange. the organic supports are really hard to remove from my prints. while normal supports respect the 'Top ocontact Z distance' and are easy to remove.
i gues this is mostly a issue for printing small items with small layer height. but it would be great if this could be changed.

closed 2026-09-29 by prusaslicer-bot[bot]