prusa3d/PrusaSlicer #9702: Unexpected Scale behavior

Agent-Cain, 2023-02-13:
### Description of the bug

If you Scale a model to lets say 250% and want to go with 200% instead it scales it based on the new 250% base. So its technically 500%  instead of 200% (as it is showing). So from 500% I have to type in 20% to go back to 100%
This wasn't like that in the 2.5 relaese and seems a bit counter intuitiv. 

### Project file & How to reproduce

I think a screenshot wouldn't show anything interessting, and also the project file doesn't give any further information.
[Little Joe_Xmas Elf_WHITE.zip](https://github.com/prusa3d/PrusaSlicer/files/10719672/Little.Joe_Xmas.Elf_WHITE.zip)

### Checklist of files included above

- [X] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.6.0-alpha3

### Operating system

Win10

### Printer model

Hypercube - dual extruder - single nozzle

kubispe1, 2023-02-13:
Hi, There is a problem that object and multipart objects behave with scale differently. That's the core of the issue. Thanks for your report.

kubispe1, 2023-02-16:
Thanks for your issue (SPE-1469), the fix will be part of the next release.