prusa3d/PrusaSlicer #9704: Mouse schemes for navigation styles

onolox, 2023-02-13:
Different schemes to mouse buttons for "navigaton styles", something like FreeCad have.
It's confusing to use different navigation styles.

The problem is that everyone uses different apps to edit in 3D, and is a needed feature to have the same mouse functionality in regards of camera movement. 

I think an option in Settings with some options to change the behavior of the mouse is the simplest approach.

kubispe1, 2023-02-13:
Thanks for FR. https://wiki.freecad.org/Mouse_navigation
I think that this option offers
- FreeCad
- Fusion 360
- New Simplify 3d

yahbluez, 2023-03-16:
seconde that

FabioM3, 2026-09-18:
If the new version don't have this option then this have to stay open.

Jan-Soustruznik, 2026-10-20:
Hello, This feature request has already been implemented in the new version of PrusaSlicer (3.0.0-alpha12). I am closing this issue as resolved. Thank you, and have a great day.

closed 2026-10-20 by Jan-Soustruznik