prusa3d/PrusaSlicer #9708: PrusaSlicer crashes when slicing overlaping objects

ymralin, 2023-02-13:
### Description of the bug

PrusaSlicer crashes when trying to slice overlaping objects.
I first encountered it when i tried to mix 2 models into one miniature. When it happened twice, i tested it with 2 simple cubes. It always crashes.

### Project file & How to reproduce

Insert any 2 shapes. Move them so they have some common volume.

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

PrusaSlicer-2.6.0-alpha3

### Operating system

Windows 10

### Printer model

Elegoo Neptune 2s

lukasmatena, 2023-02-14:
Cannot reproduce. Please, enclose a 3MF with the problematic project. Thanks.

ymralin, 2023-02-14:
I think that shapes should be merged.
Attached project crashes my slicer 100% times. Tested a couple of times after doing some other testing and after rebooting my system.
I managed to slice similar model once, but i don't know what was the exact difference between that project and attached one. I didn't save it.
[Merged.zip](https://github.com/prusa3d/PrusaSlicer/files/10734115/Merged.zip)

lukasmatena, 2023-02-15:
Thanks, reproduced this time. Most likely the same as #9679.

kubispe1, 2023-02-20:
I retested this issue and for the newly released 2.6.0-alpha4, this looks ok. In case of trouble, let us know. Thanks

closed 2023-02-20 by kubispe1