prusa3d/PrusaSlicer #9710: Missing dropdown/popup menus

diandden, 2023-02-14:
### Description of the bug

In the middle of working on a project the dropdown and right click pop up windows go blank. Instead of a list of choices the window is blank. Happens randomly. Short cuts such as alt s still work.
[image]

### Project file & How to reproduce

No can do. Issue is random.
[image]

### Checklist of files included above

- [ ] Project file
- [X] Screenshot

### Version of PrusaSlicer

Version 2.5.0+win64

### Operating system

Windows 11

### Printer model

mk3s

lukasmatena, 2023-02-14:
This is weird, I do not remember seeing this before. Can you please test this in [2.6.0-alpha3](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.6.0-alpha3)? We have updated wxWidgets in the meantime, there is a chance this might fix "itself". Thanks.

diandden, 2023-02-22:
I think the alpha is better, but it could be a Win 11 issue. Win 11 is really buggy.

 

Dennis

diandden, 2023-02-23:
I believe the problem was due to the selected Windows theme. I noticed other menus disappearing. Changing the theme seems to have resolved it.

 

Dennis

lukasmatena, 2023-11-12:
Thanks. I will close this with the assumption that it really was caused by Windows itself.

closed 2023-11-12 by lukasmatena