prusa3d/PrusaSlicer #9715: slicer changing default temp

CP0212, 2023-02-14:
### Description of the bug

I have updated my Prusa slicer and I completed some prints with no problems. Today I went to slice something and didnt even look at the temp because its always ok, but the print was looking awful. I looked at the printer and the bed was at 0. When I checked the slicer the defaults for PLA had been set to nozzle 200 bed 0. 
Why is this happening? It has done it in the past also.

### Project file & How to reproduce

[image]

### Checklist of files included above

- [ ] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.5.0

### Operating system

windows 10

### Printer model

Prusa i3mk3s+

lukasmatena, 2023-02-15:
It is hard to tell, given how much info you provided. But the default value for bed_temperature was always zero. You are apparently using the `--- default ---` profile by mistake, without any changes specific for your printer.

CP0212, 2023-02-15:
I'm sorry after I posted this I realized that I had used a profile other than the default. Now I am trying to figure out how to delete those profiles.

closed 2023-02-15 by lukasmatena