prusa3d/PrusaSlicer #9716: Variable layer height for supports and infill to speed up high detail prints

csweeney0926, 2023-02-14:
With the new organic supports and a small nozzles, printing objects like miniatures with an FDM printer is potentially feasible. Besides miniatures, I have also printed other small items using a smaller diameter nozzle and thin layer heights to get very highly detailed parts. 

This biggest downside to doing these prints is they take a lot of time (which is something new companies like Bambu are trying to address by increasing speed without compromising quality). However, I think I have a possible software adjustment that will result in significant time savings without compromising print quality or requiring any hardware changes to increase speed. The idea is basically to use different layer heights (as multiples of the thinnest, usually perimeter) layer height for infill and supports.

Let's say we are printing a highly detailed part with supports and a typical infill level. The print is using a 0.25 mm nozzle and a layer height of 0.07 mm (default ultradetail settings). Currently this would print the perimeters, infill, and support structures in layers 0.07 mm high. With variable layer height the perimeters would still print at 0.07 mm, but infill would print at a layer height of double that; 0.14 mm, and support structures would print at layer height of triple that; 0.21 mm. This will make the finish quality of the infill not as good as the perimeters (but we don't see the infill so it doesn't matter) and would halve the time it takes to print the infill. For the support structures the quality would be even worse but again we don't care because supports ultimately get thrown away anyway and these structures would get printed in ~1/3 the normal time, a significant time saving for models with lots of organic supports.

The challenges: With the variable layer heights described above the possibility of the nozzle crashing into other print layers becomes a potential issue, also we would have challenges where the print needs to come in contact with support structures. There are a couple of straightforward solutions that could be implemented. The best approach for the layers where the print and support need to come in contact with each other is probably to revert infill and support layer heights to match the perimeter layer height for the layer of and 2 layers below any layer where there is an overhang or bridge coming in contact with support material. This ensures that these supported areas print properly. For the remainder of the print, where support is not in contact with the print, simple print component order and/or z- hop increases can be used to avoid collisions. For example; perimeters are typically printed first for quality prints so two 0.07 mm layers of perimeter could be extruded before extruding a single 0.14 mm layer of infill. Then a single layer of 0.21 mm layer could be extruded for the support material structure. Using this approach and order of extrusion, the z hop only has to be increased from the default value by the height of the perimeter layer (and that is only so that the nozzle doesn't crash into a piece of support on it's way to start the next layer of perimeters).

This description has been detailed to reduce the print time of small detailed prints using a small diameter 0.25 nozzle but a similar layer multiples approach could be used to signifigantly speed up standard 0.4 mm diameter nozzle prints as well. Using the same approach as above, the perimeter layer height would be 0.1 mm (again default ultradetail), the infill layer height would be 0.2 mm, the support structure layer height would be 0.3 mm, and z- hop would increase by 0.1 mm. This would give a surface finish of the highest possible quality while having print times closer to "normal" or even "fast" default slicer settings. There are probably limits to how far this can be taken. For example applying the approach to normal settings; perimeter layer height of 0.2 mm, infill height of 0.4 mm, and support layer height of 0.6 mm would probably start to get a little sloppy but it might be worth experimenting with. Further larger 0.6 mm and 0.8 mm nozzles could see similar approaches to increase speed. 

If anyone takes an interest in this idea and thinks it has merit feel free to reach out if you want to discuss or want further clarification. I really think this has the potential to make a big increase to the potential print capacity of existing printers.

neophyl, 2023-02-14:
You can already combine infill layers to print them at thicker layer heights.  Print Settings>Infill>Reducing Print Time>Combine Infill every.

Traditional supports already use variable layer height, they are not tied to the model.  Organic is different it seems in that as yet they don't support variable height at all on the model or supports.  I'm sure that will come in due time as the devs have said they are working on it.

fer662, 2023-07-31:
Combining support layers seems like a no brainer feature, as it's a place where you don't mind losing resolution at all.

closed 2026-09-29 by prusaslicer-bot[bot]