prusa3d/PrusaSlicer #9717: End G-code for Prusa Mini is broken

pjnovas, 2023-02-14:
### Description of the bug

after I updated slicer I started to notice the end prints travel the head of the printer to a weird place. So I checked the "Custom G-Code" that comes with the config of the printer and it has `{endif}` in the comment of the line

### Project file & How to reproduce

1. Open Prusa Slicer
2. go to `Printer Settings`
3. select `Original Prusa MINI & MINI+` 
4. click on `Custom G-code`
5. review code inside `End-G-code` section 

This is the current End G-code that is coming with the software (line 2 and 4)
```g-code
G1 E-1 F2100 ; retract
{if max_layer_z < max_print_height}G1 Z{z_offset+min(max_layer_z+2, max_print_height)} F720 ; Move print head up{endif}
G1 X178 Y178 F4200 ; park print head
{if max_layer_z < max_print_height}G1 Z{z_offset+min(max_layer_z+30, max_print_height)} F720 ; Move print head further up{endif}
G4 ; wait
M104 S0 ; turn off temperature
M140 S0 ; turn off heatbed
M107 ; turn off fan
M221 S100 ; reset flow
M900 K0 ; reset LA
M84 ; disable motors
; max_layer_z = [max_layer_z]
```

This is how it should be
```g-code
G1 E-1 F2100 ; retract
{if max_layer_z < max_print_height}G1 Z{z_offset+min(max_layer_z+2, max_print_height)}{endif} F720 ; Move print head up
G1 X178 Y178 F4200 ; park print head
{if max_layer_z < max_print_height}G1 Z{z_offset+min(max_layer_z+30, max_print_height)}{endif} F720 ; Move print head further up
G4 ; wait
; removed code from here for simplicity
```

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.5.0+win64

### Operating system

Windows 10

### Printer model

Original Prusa MINI+

rtyr, 2023-02-14:
Current end g-code is correct. The behavior with "travel to the weird place" is caused by firmware bug. This has been fixed long time ago, so it seems you are using fairly old firmware on your printer. Please update your firmware.

Link to the current FW version:
https://www.prusa3d.com/downloads/firmware/prusa3d_fw_4_4_1_MINI.zip

Closing.

closed 2023-02-14 by rtyr

pjnovas, 2023-02-15:
so what you are saying is that code with an endif inside the comments section is correct?.

I'm gonna leave it here for anyone else who has the same problem then: if you move the endif code outside of the comments section the printer will actually work just fine without updating you firmware. just checked 😉

rtyr, 2023-02-15:
> o what you are saying is that code with an endif inside the comments section is correct?

Yes, it is correct. Unlike the g-code provided by you (which was actually used previously), which can result in command "F720" when condition at the beginning are not being met.

The bug in the firmware I mentioned caused that some part of the comment section at the end of g-code was falsely executed as g-code. Usually resulting in false movement after print. So any change you will do to the start g-code / end g-code can "fix it" or make it worse.

rtyr, 2023-02-15:
Also see https://github.com/prusa3d/PrusaSlicer-settings/issues/139.

and 

https://github.com/prusa3d/Prusa-Firmware-Buddy/releases/tag/v4.3.2 (last chapter "Extra head movements after the print ends")

So I really suggest you to update your FW.

> I'm gonna leave it here for anyone else who has the same problem then: if you move the endif code outside of the comments section the printer will actually work just fine without updating you firmware. just checked 😉

Please do not spread this incorrect "solution".