prusa3d/PrusaSlicer #9719: Initial Flow Rate

MahindraPersaud, 2023-02-14:
**Is your feature request related to a problem? Please describe.**
Not necessarily a problem, but on my prints I like to have an initial flowrate for the first layer to ensure great adhesion and often times use it to fix minor gaps. However, within prusaslicer, there is no `Inital Layer Flow Rate`.

**Describe the solution you'd like**
I would like a new parameter added into print settings -> Advanced -> Flow called "Inital Flow Ratio"

**Describe how it would work**
Upon entering an initial flow rate it will embed M221 S<value> into the gcode for layer 0.
This will only be used for Layer 0 and future layers would reset back to the default.

**Describe alternatives you've considered**
I have embedded the following code into my layer change G code
```
;BEFORE_LAYER_CHANGE
G92 E0.0
M221 S{if layer_num==0}110{else}95{endif}
;LayerZ: [layer_z]
;LayerNUM: [layer_num]
```

**Additional context**
I would be more than happy to build and implement it myself. I have already cloned the repo and have used the automated script to build the project. I just need a bit of direction in where I should start looking and what I should take into account etc.

filipdimitrovcom, 2024-04-10:
Hello MahindraPersaud,
Any movement on this issue? 
Seems a lot of people awaits this option :)
cheers

Nephrite-FM, 2025-10-08:
This needs to be added for users that use custom textured build plates.  The substitute code:
M221 S{if layer_num==0}110{else}95{endif}
fails to take effect on layer 0 on the latest version, so you need to start M221 in the initial G-Code of the printer first and then force it back to default in ;BEFORE_LAYER_CHANGE.

closed 2026-09-29 by prusaslicer-bot[bot]