prusa3d/PrusaSlicer #9722: 2.6.0 alpha 3 - z offset and start gcode prime line causing first layer issues

dojohnso, 2023-02-14:
### Description of the bug

When slicing with a Z Offset and start gcode with a prime line, the first layer is (first layer height) - (z offset) and only contains the prime line. layer two is at (first layer height) and contains the full first layer of the model. 

I noticed while it is generating gcode, when showing the slice preview, the first layer of the model is on layer 1, but then after gcode is generated is when the first layer shifts, hence me looking at the start gcode. If I adjust the Z in the start gcode prime strip the first layer shifts with it, no matter if it's above or below the first layer height. 

Screenshots and start gcode below. Please let me know if there is anything else I can provide. 

This happens with multiple models from different artists. If i comment out the four lines under "PRIME STRIP" below, it slices with and without z offset set. This does not happen in 2.5.

[image]
[image]
```M413 S0; disable power loss recovery
G21 ;metric values
G90 ;absolute positioning
M82 ;set extruder to absolute mode

M140 S{first_layer_bed_temperature[0]} ; get it started now

G28 ; Home all axes
G92 E0 ; Reset Extruder
G1 Z2.0 F3000 ; Move Z Axis up little to prevent scratching of Heat Bed
G1 X2 Y20 F5000.0 ; Move to start position
G1 Z0.1 F3000 ; Move Z Axis down to the bed to try and prevent oozing while it heats

M104 S{first_layer_temperature[0]} ; set final nozzle temp
M190 S{first_layer_bed_temperature[0]} ; wait for bed temp to stabilize
M109 S{first_layer_temperature[0]} ; wait for nozzle temp to stabilize

;;; PRIME STRIP

G1 Z0.3 F5000.0 ; Move to start Z position
G1 X2 Y150.0 Z0.3 F1500.0 E15 ; Draw the first line
G1 X2.4 Y150.0 Z0.3 F5000.0 ; Move to side a little
G1 X2.4 Y20 Z0.3 F1500.0 E30 ; Draw the second line

G92 E0 ; Reset Extruder
G1 Z2.0 F3000 ; Move Z Axis up little to prevent scratching of Heat Bed
G1 F2400 E-1
G1 X5 Y20 Z0.3 F5000.0 ; Move over and down to prevent blob squish```
![image](https://user-images.githubusercontent.com/294549/218857348-ac05a9d4-5916-44b1-afe2-b39eb8d5f654.png)
![image](https://user-images.githubusercontent.com/294549/218857360-f08fc15a-6599-4231-8968-a05fa3aed48b.png)

if i set the Z offset to 0 on the printer settings, the first layer is (first layer height) and contains the prime stripe and first layer of the model.
![image](https://user-images.githubusercontent.com/294549/218858920-ca514cda-baa7-4534-99ce-729d3c808fad.png)

### Project file & How to reproduce

[cinderwing3d-tinyphoenix-r2.3mf.zip](https://github.com/prusa3d/PrusaSlicer/files/10736793/cinderwing3d-tinyphoenix-r2.3mf.zip)

layer 1
![image](https://user-images.githubusercontent.com/294549/218861078-26839519-2117-4c03-a28c-03f49c9a5663.png)

layer 2
![image](https://user-images.githubusercontent.com/294549/218861104-69c431ed-d008-40f9-a744-56c4d2e0f451.png)

Just click slice, drop the preview down to the first and second layers. 

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

Version 2.6.0-alpha3+MacOS-x64

### Operating system

macOS 13.1 (22C65)

### Printer model

Anycubic Kobra / Ender 3 Pro / etc

lukasmatena, 2023-02-15:
I checked and it indeed behaves differently that 2.5.0. Apparently only a glitch in the Preview though, the beginning of the G-Code looks the same in both cases. It looks similar to #8202, but that was only happening in vase mode...

FidelCapo, 2023-02-27:
Thank you for the report.
This issue should be fixed and the fix will be included in the next release.
Closing.

closed 2023-02-27 by FidelCapo